react-native-camera/react-native-camera #2343: Barcode PDF417 working properly only when Remote JS Debugger is ON

marlonfrr, 2019-06-26:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
Yes
- [ ] Did you try master?
Yes
- [ ] Did you look for existing matching issues?
Yes

**Platforms**

Android API 26

IOS 12.2

**Versions**

- Android: api 26
- iOS: 12.2
- react-native-camera: ^2.11.0
- react-native: ^0.59.0
- react: 16.5.1
  

**Description/Current Behaviour**

I'm trying to read PDF417 barcode format within android emulator (but problem persists both physical device and simulator). The output on _onRead_ (TURNING ON REMOTE DEBUGGER) is the following:

[image]

As you can see, data contains all the information that is extracted from de ID Card just fine, but the problem comes when I turn off the Remote Debugger, the output is the following:

[image]

NOTE that the data is just the first portion of the first log

And when I test on a real iPhone the output also contains an extra "dataRaw":

[image]

**Expected Behaviour**

The data extracted should be the same on both android and iOS and without remote debugger.

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

andresmtz98, 2019-07-03:
Hey bro, i'm having this issue, please help me :(

andresmtz98, 2019-07-04:
I don't know why this happens but a workaround that i wrote after hours:
[image]

MateusAndrade, 2019-07-15:
This is a really strange bug. :open_mouth: 

@marlonfrr  you can tell more about how you managed to solve this?

@andresmtz98  thanks for providing your workaround. :tada:

andresmtz98, 2019-07-15:
> This is a really strange bug. 😮
> 
> @marlonfrr you can tell more about how you managed to solve this?
> 
> @andresmtz98 thanks for providing your workaround. 🎉

Ty bro. Another critical bug is this: #794

renatobenks, 2020-06-11:
@estevg it's probably solved in the latest, which version are you using?

estevg, 2020-06-12:
> @estevg it's probably solved in the latest version, which version are you using?

use the version react-native-camera@3.26.0 but it presents the same problem

renatobenks, 2020-06-12:
ok @estevg, I'll be checking this out!

estevg, 2020-06-24:
> ok @estevg, I'll be checking this out!

any solution?

renatobenks, 2020-07-15:
not yet @estevg, I still couldn't reproduce it, do you have a repo with such problem happening? it will be useful to fix ASAP for sure

closed 2020-09-20 by stale[bot]