react-native-camera/react-native-camera #2344: Android recordAsync is resolving without calling stopRecording()

studiobrain, 2019-06-27:
# Bug Report

**To Do First**

- [ x] Did you try latest release?
- [x ] Did you try master?
- [x ] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: version 9
- iOS:
- react-native-camera: "react-native-camera": "^2.11.0"
- react-native: "react-native": "0.59.9"
- react: "react": "16.8.3",
- react-navigation: "react-navigation": "^3.6.1"

**Description/Current Behaviour**

When calling `camera.recordAsync(options)` it is immediately resolved and there is no recording.

**Expected Behaviour**

Calling `camera.stopRecording();` woulod cause the recording to stop and resolve

```startRecordingVideo = async () => {
        const options = {
          quality: 0.5,
          fixOrientation: true,
          forceUpOrientation: true,
          videoOrientation: 1,
          deviceOrientation: 1,
          mute: true,
        };
        this.recording = true;
        console.log("recording");
        const { uri, codec } = await this.camera.recordAsync(options);
        console.log("done?"); // logs immediately after the call
        const { navigation } = this.props;

        this.media.initialize({
          uri,
          codec,
          navigation,
          type: "video",
          screen: "ChatVideoPreview",
        });
};```

```stopRecordingVideo = () => {
        console.log("stopping?"); // never logged
        if (this.recording) this.camera.stopRecording();
        this.recording = false;
};```

called via:
```let button = !this.recording ? (
          <TouchableOpacity
            activeOpacity={0.7}
            style={styles.videoButton}
            onPress={() => this.startRecordingVideo()}>
            <View style={styles.videoIcon} />
          </TouchableOpacity>
        ) : (
          <TouchableOpacity
            activeOpacity={0.7}
            style={styles.buttonRecording}
            onPress={() => this.stopRecordingVideo()}>
            <View style={styles.videoIconRecording} />
          </TouchableOpacity>
        );```

I am receiving the response data but no video and it is immediately.  Any direction and or guidance is as always appreciated, so thanks in advance!

MateusAndrade, 2019-07-01:
Hi @studiobrain , did you set the correct permission on ```AndroidManifest.xml```?

Please give a look at our [examples](https://github.com/react-native-community/react-native-camera/tree/master/examples) and let us know if this helps you. :)

studiobrain, 2019-07-02:
Thanks for the reply @MateusAndrade ! 
Here are my permissions currently:
```<uses-permission android:name="android.permission.INTERNET" />
    <uses-permission android:name="android.permission.RECORD_AUDIO"/>
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.FOREGROUND_SERVICE" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.CAMERA" />
    <uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>```

Admittedly I did just added the:
```<uses-permission android:name="android.permission.CAMERA" />
<uses-permission android:name="android.permission.SYSTEM_ALERT_WINDOW"/>```

...yet I still have intermittent immediate promise resolves.  Not sure if this is something you can take to the bank, but it seems to occur on the 3rd video taken:
`take video`
`run some other operations`
`take video`
`run some other operations`
`take video` => immediate promise resolve

Im noticing this error (YellowBox) as well:

`Possible Unhandled Promise Rejection`
```Error: unlock failed
    at createErrorFromErrorData (blob:file:///b64bd8c1-9362-4755-b408-0c306432c6ba:19195:17)
    at blob:file:///b64bd8c1-9362-4755-b408-0c306432c6ba:19147:27
    at MessageQueue.__invokeCallback (blob:file:///b64bd8c1-9362-4755-b408-0c306432c6ba:20003:18)
    at blob:file:///b64bd8c1-9362-4755-b408-0c306432c6ba:19734:18
    at MessageQueue.__guard (blob:file:///b64bd8c1-9362-4755-b408-0c306432c6ba:19907:13)
    at MessageQueue.invokeCallbackAndReturnFlushedQueue (blob:file:///b64bd8c1-9362-4755-b408-0c306432c6ba:19733:14)
    at e (file:///Applications/React%20Native%20Debugger.app/Contents/Resources/app.asar/js/RNDebuggerWorker.js:1:136705)```

...thoughts?

MarcoScabbiolo, 2019-07-31:
I've seen this happen also in iOS, couldn't find any steps to reproduce the error, but it seems to be related to the instance of the app, if you restart the app the problem is solved. When it occurs, it does for every video recorded.

I can assure you the `stopRecording` function is no being called, the `recordAsync` promise resolves on its own with an empty video.

antbui, 2019-08-02:
Hi @studiobrain,

Did you have the solution for this issue? 

Thanks,
An Bui.

closed 2019-10-08 by stale[bot]

SamiChab, 2021-02-16:
I have the same error on one Android device only:
- OnePlus 5T; Android 8.0.0
I tested on other devices and it works fine

I do not call stopRecording(), but recordAsync resolves with a video of approx. 1 second that has only one frame but has sound

Can anyone help?

mustafa-hanif, 2021-06-26:
I am also facing the same issue