react-native-camera/react-native-camera #2345: "RNCamera" was not found in UIManager

vedant17, 2019-06-28:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

What to do here?

Invariant Violation: Invariant Violation: requireNativeComponent: "RNCamera" was not found in the UIManager.

This error is located at:
    in RNCamera (at RNCamera.js:738)
    in RCTView (at View.js:45)
    in View (at RNCamera.js:737)
    in Camera (at react-native-qrcode-scanner/index.js:231)
    in RCTView (at View.js:45)
    in View (at createAnimatedComponent.js:151)
    in AnimatedComponent (at react-native-qrcode-scanner/index.js:225)
    in RCTView (at View.js:45)
    in View (at react-native-qrcode-scanner/index.js:267)
    in QRCodeScanner (at scanQRCode.js:19)
    in RCTView (at View.js:45)
    in View (at scanQRCode.js:18)
    in ScanQRCode (at SceneView.js:9)
    in SceneView (at FluidTransitioner.js:377)
    in RCTView (at View.js:45)
    in View (at createAnimatedComponent.js:151)
    in AnimatedComponent (at screens.native.js:59)
    in Screen (at TransitionRouteView.js:45)
    in TransitionRouteView (at FluidTransitioner.js:368)
    in RCTView (at View.js:45)
    in View (at screens.native.js:83)
    in ScreenContainer (at TransitionItemsView.js:119)
    in RCTView (at View.js:45)
    in View (at TransitionItemsView.js:113)
    in TransitionItemsView (at FluidTransitioner.js:223)
    in RCTView (at View.js:45)
    in View (at Transitioner.tsx:267)
    in Transitioner (at FluidTransitioner.js:73)
    in FluidTransitioner (at createFluidNavigator.js:38)
    in FluidNavigationView (at createNavigator.js:61)
    in Navigator (at createAppContainer.js:429)
    in NavigationContainer (at App.js:22)
    in App (at renderApplication.js:35)
    in RCTView (at View.js:45)
    in View (at AppContainer.js:98)
    in RCTView (at View.js:45)
    in View (at AppContainer.js:115)
    in AppContainer (at renderApplication.js:34)

This error is located at:
    in NavigationContainer (at App.js:22)
    in App (at renderApplication.js:35)
    in RCTView (at View.js:45)
    in View (at AppContainer.js:98)
    in RCTView (at View.js:45)
    in View (at AppContainer.js:115)
    in AppContainer (at renderApplication.js:34)
getNativeComponentAttributes
    getNativeComponentAttributes.js:29:4
get
    ReactNativeViewConfigRegistry.js:104:25
createInstance
    ReactNativeRenderer-dev.js:3988:53
completeWork
    ReactNativeRenderer-dev.js:14437:12
completeUnitOfWork
    ReactNativeRenderer-dev.js:17094:10
performUnitOfWork
    ReactNativeRenderer-dev.js:17304:30
workLoop
    ReactNativeRenderer-dev.js:17316:41
renderRoot
    ReactNativeRenderer-dev.js:17417:15
performWorkOnRoot
    ReactNativeRenderer-dev.js:18423:17
performWork
    ReactNativeRenderer-dev.js:18324:24
performSyncWork
    ReactNativeRenderer-dev.js:18285:14
requestWork
    ReactNativeRenderer-dev.js:18169:19
scheduleWork
    ReactNativeRenderer-dev.js:17969:16
enqueueSetState
    ReactNativeRenderer-dev.js:6934:17
setState
    react.development.js:335:31
componentDidMount$
    RNCamera.js:701:6
tryCatch
    runtime.js:45:44
invoke
    runtime.js:271:30
tryCatch
    runtime.js:45:44
invoke
    runtime.js:135:28
<unknown>
    runtime.js:145:19
tryCallOne
    core.js:37:14
<unknown>
    core.js:123:25
_callTimer
    JSTimers.js:152:14
_callImmediatesPass
    JSTimers.js:200:17
callImmediates
    JSTimers.js:473:30
callImmediates
    [native code]
__callImmediates
    MessageQueue.js:337:6
<unknown>
    MessageQueue.js:135:6
__guard
    MessageQueue.js:314:10
flushedQueue
    MessageQueue.js:134:17
flushedQueue
    [native code]
invokeCallbackAndReturnFlushedQueue
    [native code]

**Tags**

RNCamera

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

vedant17, 2019-06-28:
I'm using react-native-qrcode-scanner which uses RNCamera and I'm getting that error. It asked for permission and that error occured. What to do?

ahmad6242, 2019-07-01:
I am having this same error. I just want to capture the photos. 
Anyone can help me !

vedant17, 2019-07-01:
I solved it by uninstalling and then re installing and linking it correctly. I wasn't linking it correctly

MateusAndrade, 2019-07-01:
@ahmad6242 please you can try linking rn-camera again?

closed 2019-07-01 by MateusAndrade

emersonlaurentino, 2019-08-06:
[image]

Someone can help me?

I have the same problem and tried all the suggestions above, but none worked.

MateusAndrade, 2019-08-11:
> I have the same problem and tried all the suggestions above, but none worked.

Which version of react-native-camera and react-native are you using? Also, did you link rn-camera properly?

junaiid-khattak, 2019-11-26:
Had the same problem, rebuild fixed the problem. I have noticed that similar issues with other libs as well. I think linking doesn't properly take affect with reloading

robertoclavijob, 2019-11-27:
I  followed the steps to link react-native-camera **Manually**  on Android to get it working
https://react-native-community.github.io/react-native-camera/docs/installation#manual-install

MagicMirouff, 2019-12-02:
Same problem with autolinking. I had to setup it manually

Muhammad-Altabba, 2020-02-10:
I am using expo cli. How this linking issue could be resolved?
Thanks,

Muhammad-Altabba, 2020-02-10:
I used `expo-camera` and it worked well with `expo`.
Thanks,

creditcoder, 2020-02-14:
> I followed the steps to link react-native-camera **Manually** on Android to get it working
> https://react-native-community.github.io/react-native-camera/docs/installation#manual-install

It worked in my case, even though it's not the recommended way.

cjae, 2020-02-21:
> I followed the steps to link react-native-camera **Manually** on Android to get it working
> https://react-native-community.github.io/react-native-camera/docs/installation#manual-install

This worked.

dhirenchotwani, 2020-03-04:
> I used `expo-camera` and it worked well with `expo`.
> Thanks,

Hi, I'm also using expo cli,
can you please describe the steps in details about how to solve the linking issue ?
Thanks in advance

snakehead007, 2020-11-08:
> > I followed the steps to link react-native-camera **Manually** on Android to get it working
> > https://react-native-community.github.io/react-native-camera/docs/installation#manual-install
> 
> This worked.

404 link is dead

Tamsen1995, 2020-12-08:
any updates on this? The manual linking link is dead.

art1373, 2021-03-13:
rn camera turned to a corporate npm pkg or something? docs are super old and we have to deal with this? now still?

NicolasBonnin, 2021-07-08:
Use this link
https://react-native-camera.github.io/react-native-camera/docs/installation