react-native-camera/react-native-camera #2348: Is here someone who can actually use this library?

emildatcu, 2019-07-01:
# Question
This is retarded. I come from React, but this package management for react-native is stupid. I did all the steps written in the docs and I still get errors. Each time I fix an error, another 10 pop up. This is not how it's supposed to work. I understand that there can be various bugs which will be fixed in time, but I CAN'T USE THIS useless package and it's freaking frustrating...

I'm trying to use it on Ubuntu 18.04. 
React 16.8.3
React-native 0.59.9
React-native-camera 2.11.0

I even tried running the examples from this repo, same issues. Is there anyone who has a working version of this?

sibelius, 2019-07-01:
what error?

dhee9000, 2019-07-01:
There's probably plenty of people who are using this package as npm and the number of contributors shows but as a new user of this package (I'm trying to migrate away from Expo Camera) I'm having a good amount of trouble getting set up as well.

I'm starting off with trying to get things working on Android and after following the steps in the installation guide (which were confusing to follow, unlike most other react native packages I've seen) I keep encountering
``` 
Error while updating property 'exposure' of a view managed by:RNCamera

null

No virtual method setExposureCompensation(I)V in class Lorg/reactnative/camera/RNCameraView; or its super classes (declaration of 'org.reactnative.camera.RNCameraView' appears in /data/app/co.ripplify.app-ksNe5YjMxeffYdPTfGllqQ==/base.apl!classes131.dex)
```

I checked for this method and it was declared so I'm wondering if I maybe missed a step during installation or misconfigured something.

Steps I Followed:
1. `yarn add react-native-camera`
2. `react-native link react-native-camera`
3. Edit android/build.gradle to include `missingDimensionStrategy 'react-native-camera', 'general'` since it looks like the link script doesn't add this.
4. Change `import {Camera} from 'expo-camera';` to `import {RNCamera as Camera} from 'react-native-camera';`
5. `react-native run-android`

react 16.8.3
react native 0.59.8
react-native-camera 2.11.0

Platform: Android API 28 (9 Pie)
Device: Samsung Galaxy S9+

sibelius, 2019-07-01:
check docs https://react-native-community.github.io/react-native-camera/ and help us improve it

dhee9000, 2019-07-01:
I'd be happy to do so and submit a PR once I can resolve this issue. Is this something you have seen before? I couldn't find any issues or references to it anywhere else.

hustlercoder, 2019-07-01:
Yeah i use it....you need to follow step by step the procedure to install it and link it and after that check the example directory and try to implement your own.

closed 2019-07-01 by sibelius