react-native-camera/react-native-camera #2349: react-native-camera iOS is not detecting any faces

St7-07, 2019-07-01:
# Bug Report

**To Do First**

- [ YES ] Did you try latest release?
- [ YES ] Did you try master?
- [ YES ] Did you look for existing matching issues?

**Platforms**

**Versions**

- iOS: 12.3.1
- react-native-camera: ^2.11.0
- react-native: ^0.59.9
- react: ^16.8.6

**Description/Current Behaviour**

onFacesDetected is nos detecting any face, just return empty array. 
logs: Object {type: "face", faces: Array(0), target: 259}

Text Detector is working
onFaceDetectionError is never triggered 

**Expected Behaviour**

Detect faces when there is faces in the camera and return a filled array

**Steps to Reproduce**

Install react-native-camera via cocoa pods like 

  pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
    'FaceDetectorMLKit', 'TextDetector'
  ]

**Additionals**

Code example:

      <RNCamera
        ref={(ref) => {
          this.camera = ref;
        }}
        captureAudio={false}
        style={styles.preview}
        type={cameraType} // already probe with back and front, any of both works
        defaultVideoQuality={RNCamera.Constants.VideoQuality['720p']}
        permissionDialogTitle={i18n.t('KYCValidation_CameraAccessTitle')}
        permissionDialogMessage={i18n.t('KYCValidation_CameraAccessMessage')}
        onTextRecognized={({ textBlocks }) => console.log(textBlocks)}
        faceDetectionMode={RNCamera.Constants.FaceDetection.Mode.accurate} // fast doesn't work neither
        faceDetectionClassifications={RNCamera.Constants.FaceDetection.Classifications.none}
        onFacesDetected={faces => console.log(faces)}
        onFaceDetectionError={error => console.log('FDError', error)} // This is never triggered
      >

eduardinni, 2019-07-13:
same problem with the same env

demianp, 2019-07-18:
try this workaround:
onCameraReady={() => this.setState({canDetectFaces: true})}
onFacesDetected={this.state.canDetectFaces ? this.facesDetected : null}

barnapcristian, 2019-07-18:
Hello GitHub, this is my first time answering ever here :)

Here's how I managed to make it work on iOS with React Native 0.60 with both cameras front and back. Although I'm still waiting on an official fix, maybe with my answer they can pin point the problem and fix it. So for now it's temporary fix.
The problem as far as I investigated relies in these 2 lines:

    CGRect croppedSize = AVMakeRectWithAspectRatioInsideRect(previewSize, boundingRect);
    CGImageRef croppedCGImage = CGImageCreateWithImageInRect(videoImage, croppedSize);

So the solution that I found is:

1. Open Xcode with your project
2. Find and open the linked React-Native-Camera folder
3. Find and open RNCameraUtils.m
[image]

4. At the end of the file you'll find
    videoImage = [temporaryContext createCGImage:ciImage fromRect:boundingRect];
    CGRect croppedSize = AVMakeRectWithAspectRatioInsideRect(previewSize, boundingRect);
    CGImageRef croppedCGImage = CGImageCreateWithImageInRect(videoImage, croppedSize);
    UIImage *image = [[UIImage alloc] initWithCGImage: **croppedCGImage**];
    CGImageRelease(videoImage);
    CGImageRelease(croppedCGImage);
    return image;
5. Replace it with
    videoImage = [temporaryContext createCGImage:ciImage fromRect:boundingRect];
    CGRect croppedSize = AVMakeRectWithAspectRatioInsideRect(previewSize, boundingRect);
    CGImageRef croppedCGImage = CGImageCreateWithImageInRect(videoImage, croppedSize);
    UIImage *image = [[UIImage alloc] initWithCGImage:**videoImage**];
    CGImageRelease(videoImage);
    CGImageRelease(croppedCGImage);
    return image;
[image]

The problem seems to be the croppedCGImage. So if we change the **initWithCGImage:croppedCGImage**
                             with
**initWithCGImage:videoImage** 
                         it works!

**But keep in mind that it's a temporary fix and if you update RNC, this fix will disappear. So better to not update until RNC says it's fixed.** 

Have a great day!

sibelius, 2019-09-16:
@barnapcristian  can you send your fix as a pull request?

jordangrant, 2019-11-22:
@barnapcristian solution works (thanks so much) saved me time

igor-bv, 2019-11-28:
Will someone provide a PR to fix this bug and merge into original repo?

sibelius, 2019-11-29:
fixed here https://github.com/react-native-community/react-native-camera/pull/2611

closed 2019-11-29 by sibelius

viimauer, 2019-12-02:
> fixed here #2611

Its working only in front camera for me

sibelius, 2019-12-02:
@viimauer can you open a new issue for back camera?

can you dig in the code?