react-native-camera/react-native-camera #2350: Camera view threw an error - component could not be rendered.

AlekseyPimenov, 2019-07-02:
# Bug Report
- [ ] Did you try latest release? - YES
- [ ] Did you try master? - YES
- [ ] Did you look for existing matching issues? - YES

**Platforms**
Android 7.0

- Android: 7.0
- iOS: ---
- react-native-camera:2.11.0
- react-native:0.59.9
- react:16.8.3
  

**Description/Current Behaviour**
Add react-native-camera to react-native project (after init for example)
Run app on device - get error:
onMountError ->  { message: 'Camera view threw an error - component could not be rendered.' }
Get black screen and crash

Related only for Android 7.0 - 7.1
Other ones work good.

**Expected Behaviour**
react-native-camera worked on Android 7.0 -7 .1

**Steps to Reproduce**
1) Add react-native-camera to react-native project (after init for example)
2) Run app on real device with Android 7.0
3) Get black screen and crash

MateusAndrade, 2019-07-15:
Did you asked for the permissions on Android/IOS, setting AndroidManifest.xml and Info.plist?

JuAlexandre, 2019-07-30:
I've the same error, for me it appears when I make a video call with webrtc.
Once I come back to my scan screen the camera is in error.

closed 2019-10-05 by stale[bot]

dacre-denny, 2020-09-09:
To resolve this I had to uninstall and reinstall an app that was using this library