react-native-camera/react-native-camera #2353: clean install gives "Task :react-native-camera:compileGeneralDebugJavaWithJavac FAILED"

Esshahn, 2019-07-05:
# Bug Report

**To Do First**

- [x ] Did you try latest release?
- [x ] Did you try master?
- [x ] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:

**Description/Current Behaviour**

I set up a new project with nothing else than react-native-camera and running it fails with errors (other projects build fine)

**Expected Behaviour**

Expecting a successful build

**Steps to Reproduce**

````
react-native init AwesomeProject
cd AwesomeProject
npm install react-native-camera --save
react-native link react-native-camera (tried with and without)
react-native run-android
````

result:

````
C65:pansennase esshahn$ react-native run-android
error React Native CLI uses autolinking for native dependencies, but the following modules are linked manually: 
  - react-native-camera (to unlink run: "react-native unlink react-native-camera")
This is likely happening when upgrading React Native from below 0.60 to 0.60 or above. Going forward, you can unlink this dependency via "react-native unlink <dependency>" and it will be included in your app automatically. If a library isn't compatible with autolinking, disregard this message and notify the library maintainers.
Read more about autolinking: https://github.com/react-native-community/cli/blob/master/docs/autolinking.md
info JS server already running.
info Installing the app...

> Task :react-native-camera:compileGeneralDebugJavaWithJavac FAILED

Deprecated Gradle features were used in this build, making it incompatible with Gradle 6.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/5.4.1/userguide/command_line_interface.html#sec:command_line_warnings
14 actionable tasks: 2 executed, 12 up-to-date
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/RNCameraView.java:10: error: package android.support.v4.content does not exist
import android.support.v4.content.ContextCompat;
                                 ^
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:27: error: package android.support.annotation does not exist
import android.support.annotation.IntDef;
                                 ^
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:28: error: package android.support.annotation does not exist
import android.support.annotation.NonNull;
                                 ^
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:29: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
                                 ^
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:30: error: cannot find symbol
import android.support.v4.os.ParcelableCompat;
                            ^
  symbol:   class ParcelableCompat
  location: package android.support.v4.os
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:31: error: cannot find symbol
import android.support.v4.os.ParcelableCompatCreatorCallbacks;
                            ^
  symbol:   class ParcelableCompatCreatorCallbacks
  location: package android.support.v4.os
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:32: error: package android.support.v4.view does not exist
import android.support.v4.view.ViewCompat;
                              ^
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/AspectRatio.java:21: error: package android.support.annotation does not exist
import android.support.annotation.NonNull;
                                 ^
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/AspectRatio.java:22: error: package android.support.v4.util does not exist
import android.support.v4.util.SparseArrayCompat;
                              ^
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/Size.java:21: error: package android.support.annotation does not exist
import android.support.annotation.NonNull;
                                 ^
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/AspectRatio.java:29: error: cannot find symbol
    private final static SparseArrayCompat<SparseArrayCompat<AspectRatio>> sCache
                         ^
  symbol:   class SparseArrayCompat
  location: class AspectRatio
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/AspectRatio.java:29: error: cannot find symbol
    private final static SparseArrayCompat<SparseArrayCompat<AspectRatio>> sCache
                                           ^
  symbol:   class SparseArrayCompat
  location: class AspectRatio
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/tasks/ResolveTakenPictureAsyncTask.java:9: error: package android.support.media does not exist
import android.support.media.ExifInterface;
                            ^
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraViewManager.java:3: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
                                 ^
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/RNCameraViewHelper.java:9: error: package android.support.media does not exist
import android.support.media.ExifInterface;
                            ^
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/RNCameraViewHelper.java:282: error: cannot find symbol
  public static WritableMap getExifData(ExifInterface exifInterface) {
                                        ^
  symbol:   class ExifInterface
  location: class RNCameraViewHelper
/Users/esshahn/github/pansennase/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/RNCameraViewHelper.java:312: error: cannot find symbol
  public static void setExifData(ExifInterface exifInterface, WritableMap exifMap) {
                                 ^

(rest is cut off, more than 10 pages in terminal)
````

Also added in android/app/src/main/AndroidManifest.xml

````
<uses-permission android:name="android.permission.CAMERA" />
<uses-permission android:name="android.permission.RECORD_AUDIO"/>
<uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
<uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />
````

and in android/app/build.gradle

````
defaultConfig {
         applicationId "com.cameraexample"
         minSdkVersion rootProject.ext.minSdkVersion
+        missingDimensionStrategy 'react-native-camera', 'general'
         targetSdkVersion rootProject.ext.targetSdkVersion
         versionCode 1
         versionName "1.0"
````

Any help appreciated.

Esshahn, 2019-07-05:
I've followed this exact tutorial, step by step, for comparison:
https://www.toptal.com/react-native/react-native-camera-tutorial

thessler27, 2019-07-05:
@Esshahn this is because RN 0.60 [supports android X](https://github.com/facebook/react-native/releases/tag/v0.60.0). If you downgrade to RN 0.59, your install should work

Esshahn, 2019-07-05:
thank you @thessler27 for your support, I will try that. Is there no way the latest RN 0.60 can be used? Cheers!

Esshahn, 2019-07-06:
I did the same with 
````
react-native-cli: 2.0.1
react-native: 0.59.9
````
and still get errors with a clean install:
````
FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:preDebugBuild'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve project :react-native-camera.
     Required by:
         project :app
      > Cannot choose between the following variants of project :react-native-camera:
          - generalDebugRuntimeElements
          - mlkitDebugRuntimeElements
        All of them match the consumer attributes:
          - Variant 'generalDebugRuntimeElements' capability awesomeapp:react-native-camera:unspecified:
              - Unmatched attributes:
                  - Found com.android.build.api.attributes.VariantAttr 'generalDebug' but wasn't required.
                  - Found react-native-camera 'general' but wasn't required.
              - Compatible attributes:
                  - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
                  - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
                  - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.
          - Variant 'mlkitDebugRuntimeElements' capability awesomeapp:react-native-camera:unspecified:
              - Unmatched attributes:
                  - Found com.android.build.api.attributes.VariantAttr 'mlkitDebug' but wasn't required.
                  - Found react-native-camera 'mlkit' but wasn't required.
              - Compatible attributes:
                  - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
                  - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
                  - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 0s
error Could not install the app on the device, read the error above for details.
Make sure you have an Android emulator running or a device connected and have
set up your Android development environment:
https://facebook.github.io/react-native/docs/getting-started.html
error Command failed: ./gradlew app:installDebug. Run CLI with --verbose flag for more details.
````

Any ideas?

closed 2019-07-06 by Esshahn

thessler27, 2019-07-06:
@Esshahn  yup no problem. And yeah 0.60 is going to cause a lot of issues, but there [is a PR up](https://github.com/react-native-community/react-native-camera/pull/2306) that would fix this, but then would not be backwards compatible with <=0.59 so it'll likely be slow moving. People say that [jettifier](https://www.npmjs.com/package/jetifier) works, but I've had no success in the past. It'll probably be a slow move towards AndroidX support in the community

BaymanVova, 2019-07-09:
How to downgrade to RN 0.59?

Esshahn, 2019-07-09:
You can either do a clean install by 
````
react-native init newproject --version react-native@0.59.9
````
or change the rect version in your package.json and do an install with npm or yarn.

gentle-shao, 2019-07-15:
Hi,I just faced same problem and solved it on React-Native v0.6,here's my solution:

1. JUST FOLLOWING THIS [TUTORIAL](https://www.toptal.com/react-native/react-native-camera-tutorial)
2. EXECUTE THIS `react-native run-android --no-jetifier`
3. CHANGE YOUR App.js to this:
```import { RNCamera } from 'react-native-camera';
import React, from 'react';
import {
  View,
  Text,
} from 'react-native';

const App = () => {
  return (
    <View style={{flex: 1}}>
      <RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          style={{
            flex: 1,
            width: '100%',
          }}
        >
        </RNCamera>
    </View>
  );
};
export default App;
```

WORKS FOR ME WITHOUT DOWNGRADE RN.

wangjin1116, 2019-07-15:
谢谢您

发自我的iPhone

RyanThomas73, 2019-08-01:
It is possible to adjust the build.gradle config so that it will work for either android.support or androidx library (the consumer would still need to use jetifier to adjust the import statements). I've seen it done a couple different ways. `react-native-device-info` is a good example:
https://github.com/react-native-community/react-native-device-info/blob/master/android/build.gradle

The gist of it is:

1. In the `build.gradle`, check for a variable (with a default fallback) in the root project to determine whether to use the legacy android.support or the androidx library. Check for variables for the individual dependency versions (with fallbacks).

2. Update the `implementation xxx` statements to use variables for both the dependency name and the dependency version. The dependency name variable will be set conditionally to either the legacy `android.support` or `androidx` equivalent.

3. If necessary, adjust your import statements to match whichever library you consider the default. e.g. if you default to using androidx update your import statements / references to use the new androidx equivalent and the consumer can use jetifier in reverse mode to switch them back if they want to continue using the legacy library and react-native <= 0.59.x

if you default to using android.support leave your import statements unchanged and the consumer can use jetifier to update them to the androidx equivalent(s)

SummerJoan3, 2019-10-10:
yarn add jetifier -D
npx jetifier

hey68you, 2021-03-04:
On MacOSX bigsur, for me the problem was really that I had 2 JDK's installed:

```
$ /usr/libexec/java_home -V | grep jdk

Matching Java Virtual Machines (2):
    1.8.281.09 (x86_64) "Oracle Corporation" - "Java" /Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home
    1.8.0_282 (x86_64) "AdoptOpenJDK" - "AdoptOpenJDK 8" /Library/Java/JavaVirtualMachines/adoptopenjdk-8.jdk/Contents/Home
```

React Native needs the adoptopenjdk version, so I added that to my ~/.bash_profile the adoptopenjdk path as the **JAVA_HOME** environment variable:
    

```export ANDROID_HOME=$HOME/Library/Android/sdk
export PATH=$PATH:$ANDROID_HOME/emulator
export PATH=$PATH:$ANDROID_HOME/tools
export PATH=$PATH:$ANDROID_HOME/tools/bin
export PATH=$PATH:$ANDROID_HOME/platform-tools
export JAVA_HOME="/Library/Java/JavaVirtualMachines/adoptopenjdk-8.jdk/Contents/Home"```