react-native-camera/react-native-camera #2355: Text recognition using mlkit on ios

almazpdf, 2019-07-07:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

almazpdf, 2019-07-07:
RNCamera onTextRecognized only returns empty [] ... for some reason Not sure why? Can anyone help. Happy to share my code here

MateusAndrade, 2019-07-07:
#1446

MateusAndrade, 2019-07-07:
@almazpdf did you follow the link instructions related to Mlkit? Also, you can try the example with Mlkit to see If It works for you?

almazpdf, 2019-07-07:
Thank you for the prompt response. I tried the example before and that
worked well for me. Based on the rich feature set I decided to use mlkit. I
am successfully able to scan barcodes but text detection is not working. Is
there specific link step for text? Or is it same for mlkit?