react-native-camera/react-native-camera #2357: RN Camera Not Linking Correctly

mreorhan, 2019-07-08:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android: 7
- iOS:
- react-native-camera: git+https://git@github.com/react-native-community/react-native-camera.git
- react-native:0.59.9
- react:16.8.3
  

**Description/Current Behaviour**
FAILURE: Build failed with an exception.

* What went wrong:
Package not working correctly.
Could not determine the dependencies of task ':app:preDebugBuild'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve project :react-native-camera.
     Required by:
         project :app
      > Cannot choose between the following variants of project :react-native-camera:
          - generalDebugRuntimeElements
          - mlkitDebugRuntimeElements
        All of them match the consumer attributes:
          - Variant 'generalDebugRuntimeElements':
              - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
              - Found com.android.build.api.attributes.VariantAttr 'generalDebug' but wasn't required.
              - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
              - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.
              - Found react-native-camera 'general' but wasn't required.
          - Variant 'mlkitDebugRuntimeElements':
              - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
              - Found com.android.build.api.attributes.VariantAttr 'mlkitDebug' but wasn't required.
              - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
              - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.
              - Found react-native-camera 'mlkit' but wasn't required.

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

aidanw, 2019-07-08:
This closed issue may hold the info you need.

https://github.com/react-native-community/react-native-camera/issues/2150#issuecomment-474098081

sibelius, 2019-08-01:
use autolinking with v3

closed 2019-08-01 by sibelius