react-native-camera/react-native-camera #2362: how to solve this problem

tljackyi, 2019-07-09:
Loading dependency graph, done.
error: bundling failed: Error: While trying to resolve module `react-native-camera` from file `/Users/pilipa/Project/custapp/app/view/code/QRScanner.js`, the package `/Users/pilipa/Project/custapp/ios/Pods/react-native-camera/package.json` was successfully found. However, this package itself specifies a `main` module field that could not be resolved (`/Users/pilipa/Project/custapp/ios/Pods/react-native-camera/src/index.js`. Indeed, none of these files exist:

  * `/Users/pilipa/Project/custapp/ios/Pods/react-native-camera/src/index.js(.native||.ios.js|.native.js|.js|.ios.json|.native.json|.json|.ios.ts|.native.ts|.ts|.ios.tsx|.native.tsx|.tsx)`

closed 2019-07-10 by tljackyi

mojizo, 2019-12-24:
Hello! How did you solve the problem?

Bogdan0810, 2020-08-10:
Hey @tljackyi 

Please let me know how you solved the problem?
I got exact same issue.