react-native-camera/react-native-camera #2363: Front camera unable to barcode scan

skyride99, 2019-07-09:
# Bug Report

**To Do First**

- [x ] Did you try latest release?
- [ ] Did you try master?
- [ x] Did you look for existing matching issues?

**Platforms**

 

**Versions**

- Android: oreo
- iOS: ios 12
- react-native-camera: 2.10.2, 2.11.0
- react-native: 0.57.4
- react: 16.6.0-alpha.8af6728
  

**Description/Current Behaviour**

Trying to create an app that reads an employee badge barcode from a front camera.
 If using google mlkit or regular barcode scan from the front camera, it does not read any barcode.

**Expected Behaviour**

Use front camera of phone or tablet to read a barcode

**Steps to Reproduce**

<View style={styles.container}>
                <RNCamera
                    ref={ref => {
                        this.camera = ref;
                   }}
                    style={styles.preview}
                    ref={cam => this.camera = cam}
                    type={RNCamera.Constants.Type.front}
                    defaultTouchToFocus
                    onGoogleVisionBarcodesDetected={this.barcodeRecognized}
                    googleVisionBarcodeType={RNCamera.Constants.GoogleVisionBarcodeDetection.BarcodeType.CODE_128}
                    googleVisionBarcodeMode={RNCamera.Constants.GoogleVisionBarcodeDetection.BarcodeMode.ALTERNATE}

                >
                    <TouchableOpacity
                        style={{ backgroundColor: '#2E79F5', padding: 8, margin: 20, width: 300, height: 50}}
                        onPress={()=> this.props.navigation.navigate('Home')}
                    > 
                        <Text style={{alignSelf: 'center', color: 'white', fontSize: 18}}>CANCEL</Text>
                    </TouchableOpacity>
                </RNCamera>
            
            </View>

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

skyride99, 2019-07-09:
I got the mlkit google to work, by making the barcode bigger for the front facing camera.

skyride99, 2019-07-09:
the default barcode scanner will not work on the front

rishiankush, 2020-07-17:
@skyride99 can you please provide a proper reason why this cannot work. I mean as per @renatobenks , he told because  Google MLKit limitation. But I did not find any solid content or reason that explains its not possible.

rishiankush, 2020-07-17:
This is my code to scan from front camera that does not work:
```
import QRCodeScanner from 'react-native-qrcode-scanner';
import { RNCamera } from 'react-native-camera';
<QRCodeScanner
          onRead={this.onSuccess}
          flashMode={RNCamera.Constants.FlashMode.auto}
          cameraType={'front'}
          cameraStyle={{height:200,width:300,alignSelf:'center'}}
        />
```

Any other alternative that I can use front camera to scan QR code?

rishiankush, 2020-07-18:
So I did not find any solution for this module in react native. And I had to use Flutter to implement this module. I have to trust on Flutter from now. its way far better than react-native.

JosephSemrai, 2020-10-13:
Also having this issue.

closed 2021-01-13 by stale[bot]