react-native-camera/react-native-camera #2364: SalesforceMobileSDK-ReactNative - Task :react-native-camera:compileGeneralDebugJavaWithJavac FAILED

oagostinho, 2019-07-09:
# Bug Report

**To Do First**

- [ X] Did you try latest release?
- [ X] Did you try master?
- [X ] Did you look for existing matching issues?

**Platforms**

 Issue is here - Compilation error, when using **forcedotcom/SalesforceMobileSDK-ReactNative**

**Versions**

React Native Environment Info:
    System:
      OS: macOS 10.14.5
      CPU: (4) x64 Intel(R) Core(TM) m7-6Y75 CPU @ 1.20GHz
      Memory: 155.54 MB / 8.00 GB
      Shell: 3.2.57 - /bin/bash
    Binaries:
      Node: 12.3.1 - /usr/local/bin/node
      Yarn: 1.16.0 - /usr/local/bin/yarn
      npm: 6.9.0 - /usr/local/bin/npm
      Watchman: 4.9.0 - /usr/local/bin/watchman
    SDKs:
      iOS SDK:
        Platforms: iOS 12.2, macOS 10.14, tvOS 12.2, watchOS 5.2
      Android SDK:
        API Levels: 21, 22, 23, 24, 25, 26, 27, 28
        Build Tools: 23.0.1, 27.0.3, 28.0.3
        System Images: android-21 | Google APIs Intel x86 Atom_64, android-25 | Google APIs Intel x86 Atom_64, android-26 | Google Play Intel x86 Atom, android-27 | Google APIs Intel x86 Atom
    IDEs:
      Android Studio: 3.4 AI-183.6156.11.34.5522156
      Xcode: 10.2.1/10E1001 - /usr/bin/xcodebuild
    npmPackages:
      react: 16.4.1 => 16.4.1
      react-native: ^0.56.1 => 0.56.1
    npmGlobalPackages:
      react-native-cli: 2.0.1

This is what I get when trying to compile:
```

> Task :react-native-camera:compileGeneralDebugJavaWithJavac
/Users/Orlando/myWorkSpace2019/myproject2019/stellaxius_expenses_salesforce_react_native/StellaxiusExpensesAppReact/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/RNCameraView.java:10: error: package android.support.v4.content does not exist
import android.support.v4.content.ContextCompat;
                                 ^
/Users/Orlando/myWorkSpace2019/myproject2019/stellaxius_expenses_salesforce_react_native/StellaxiusExpensesAppReact/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:27: error: package android.support.annotation does not exist
import android.support.annotation.IntDef;
                                 ^
/Users/Orlando/myWorkSpace2019/myproject2019/stellaxius_expenses_salesforce_react_native/StellaxiusExpensesAppReact/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:28: error: package android.support.annotation does not exist
import android.support.annotation.NonNull;
                                 ^
/Users/Orlando/myWorkSpace2019/myproject2019/stellaxius_expenses_salesforce_react_native/StellaxiusExpensesAppReact/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:29: error: package android.support.annotation does not exist
import android.support.annotation.Nullable;
                                 ^
/Users/Orlando/myWorkSpace2019/myproject2019/stellaxius_expenses_salesforce_react_native/StellaxiusExpensesAppReact/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:30: error: cannot find symbol
import android.support.v4.os.ParcelableCompat;

.....
Users/Orlando/myWorkSpace2019/myproject2019/stellaxius_expenses_salesforce_react_native/StellaxiusExpensesAppReact/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/AspectRatio.java:22: error: package android.support.v4.util does not exist
import android.support.v4.util.SparseArrayCompat;
                              ^
/Users/Orlando/myWorkSpace2019/myproject2019/stellaxius_expenses_salesforce_react_native/StellaxiusExpensesAppReact/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/Size.java:21: error: package android.support.annotation does not exist
import android.support.annotation.NonNull;
                                 ^
/Users/Orlando/myWorkSpace2019/myproject2019/stellaxius_expenses_salesforce_react_native/StellaxiusExpensesAppReact/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/AspectRatio.java:29: error: cannot find symbol
    private final static SparseArrayCompat<SparseArrayCompat<AspectRatio>> sCache

/Users/Orlando/myWorkSpace2019/myproject2019/stellaxius_expenses_salesforce_react_native/StellaxiusExpensesAppReact/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/RNCameraViewHelper.java:282: error: cannot find symbol
  public static WritableMap getExifData(ExifInterface exifInterface) {
                                        ^
  symbol:   class ExifInterface
  location: class RNCameraViewHelper
/Users/Orlando/myWorkSpace2019/myproject2019/stellaxius_expenses_salesforce_react_native/StellaxiusExpensesAppReact/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/RNCameraViewHelper.java:312: error: cannot find symbol
  public static void setExifData(ExifInterface exifInterface, WritableMap exifMap) {
```

 Install react-native with forcedotcom/SalesforceMobileSDK-ReactNative (ios,android) and then install react-native-camera. After that, the command 

> react-native run-android 

is not run normally. I am getting compilation errors.

I have this in the file package.json

{
    "name": "Demo",
    "version": "0.0.1",
    "private": true,
    "scripts": {
        "start": "node node_modules/react-native/local-cli/cli.js start",
        "test": "jest"
    },
    "sdkDependencies": {
        "SalesforceMobileSDK-Android": "https://github.com/forcedotcom/SalesforceMobileSDK-Android.git#v7.1.2",
        "SalesforceMobileSDK-iOS": "https://github.com/forcedotcom/SalesforceMobileSDK-iOS.git#v7.1.2"
    },
    "dependencies": {
        "axios": "0.19.0",
        "eslint": "^4.0.0",
        "react": "16.4.1",
        "react-native": "^0.56.1",
        "react-native-camera": "2.11.0",
        "react-native-floating-action": "1.17.0",
        "react-native-force": "https://github.com/forcedotcom/SalesforceMobileSDK-ReactNative.git#v7.1.2",
        "react-native-keyboard-aware-scroll-view": "0.8.0",
        "react-native-picker-select": "6.2.0",
        "react-native-vector-icons": "6.6.0",
        "react-navigation": "^2.17.0",
        "whatwg-fetch": "1.1.1"
    },
    "devDependencies": {
        "babel-preset-react-native": "5.0.2",
        "rimraf": "2.6.2"
    }
}

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

wangjin1116, 2019-07-10:
I try this, and it works https://github.com/react-native-community/react-native-camera/issues/2353

oagostinho, 2019-07-10:
Hi @wangjin1116 

Thanks for your supporting. However, I got a response from the salesforce mobile react SDK Team that they said is not possible...I have to wait for the next version of Salesforce Mobile React SDK.

The response is in the following link = > [(https://github.com/forcedotcom/SalesforceMobileSDK-ReactNative/issues/134)]

What do you think?

Best Regards,
Orlando Agostinho

sibelius, 2019-08-01:
use autolinking with v3

closed 2019-08-01 by sibelius