react-native-camera/react-native-camera #2371: Ratio is not respected on android

herokuapp88, 2019-07-12:
# Question
I am trying to use this library to record video, but I have issues to set the custom preview size, I want to use ratio "16:9" the same as youtube uses, I used this code:

`import React from 'react';
import { Platform } from 'react-native';
import { RNCamera } from 'react-native-camera';

const DESIRED_RATIO = "16:9";

class MyCameraComponent extends Component {
    state = {}

    prepareRatio = async () => {
        if (Platform.OS === 'android' && this.cam) {
             const ratios = await this.cam.getSupportedRatiosAsync();

             // See if the current device has your desired ratio, otherwise get the maximum supported one
             // Usually the last element of "ratios" is the maximum supported ratio
             const ratio = ratios.find((ratio) => ratio === DESIRED_RATIO) || ratios[ratios.length - 1];
             
             this.setState({ ratio });
        }
    }

    render() {
        return (
            <RNCamera
                ref={(cam) => this.cam = cam}
                style={{
                  flex: 0.5,
                  justifyContent: 'space-between',
                }}
                onCameraReady={this.prepareRatio} // You can only get the supported ratios when the camera is mounted
                ratio={this.state.ratio}
            />
        );
}
}`

But this code does not show camera preview without the style setting, only after I added the style={{flex:0.5}}, the camera preview shows up.

My android device supports 3:2, 4:3, 9:5, 16:9 etc, but the preview window always stays the same and takes half of the screen, so the preview window size is actually controlled by the style instead of the ratio, is there a way to set the preview size to my custom size? how the ratio works with the preview size? can I set a custom preview size as the ratio of 16:9?

Thanks

**Tags**
[https://github.com/react-native-community/react-native-camera/issues/1280](url)

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

info 
  React Native Environment Info:
    System:
      OS: macOS 10.14.5
      CPU: (8) x64 Intel(R) Core(TM) i7-7820HQ CPU @ 2.90GHz
      Memory: 1.02 GB / 16.00 GB
      Shell: 3.2.57 - /bin/bash
    Binaries:
      Node: 11.4.0 - /usr/local/bin/node
      npm: 6.4.1 - /usr/local/bin/npm
      Watchman: 4.9.0 - /usr/local/bin/watchman
    SDKs:
      iOS SDK:
        Platforms: iOS 12.2, macOS 10.14, tvOS 12.2, watchOS 5.2
      Android SDK:
        API Levels: 23, 25, 26, 27, 28
        Build Tools: 27.0.3, 28.0.2, 28.0.3
        System Images: android-22 | Google APIs Intel x86 Atom, android-27 | Google APIs Intel x86 Atom, android-27 | Google Play Intel x86 Atom, android-28 | Google APIs Intel x86 Atom, android-28 | Google Play Intel x86 Atom
    IDEs:
      Android Studio: 3.1 AI-173.4907809
      Xcode: 10.2.1/10E1001 - /usr/bin/xcodebuild
    npmPackages:
      react: ^16.8.6 => 16.8.6 
      react-native: ^0.59.8 => 0.59.8 
    npmGlobalPackages:
      react-native-cli: 2.0.1
      react-native-cn-richtext-editor: 1.0.33
      react-native-git-upgrade: 0.2.7

bramski, 2019-07-16:
Did you find a solution? I have the same problem.

herokuapp88, 2019-07-17:
No, still I can not get custom size of the video, I tried to set ratio=1:1, but the resulting captured video is a always portrait, I want either square or landscape.

billouboq, 2019-07-29:
same here, not working, and can't find anything useful about this problem

closed 2019-10-04 by stale[bot]

KristineTrona, 2020-02-04:
This issue should be reopened, having the same issue.

Luan-Sens, 2020-03-27:
Hello community.

Apparently i have the same problem. i attached my code for example.

`export default class Example extends Component {
    
getHeightOfWidthByRatio(ratio:string){
	const screenWidth = Math.round(Dimensions.get('window').width);
	let [ratioWidth, ratioHeight] = ratio.split(':');
	let ratioCoefient = Number(ratioWidth) / Number(ratioHeight);
	return Math.round(screenWidth / ratioCoefient);
}
render(){
	return(
		<SafeAreaView style={{flex:1}}>
			<RNCamera
				style={{width:'100%',height:this.getHeightOfWidthByRatio('4:3')}}
				captureAudio={false}
				type={RNCamera.Constants.Type.back}
				ratio='4:3'
				onGoogleVisionBarcodesDetected={(data, rawData, type, bounds) => {console.log('barcode: ',data.data);}}
				playSoundOnCapture={true}
				androidCameraPermissionOptions={{
					title: 'Permissão para usar a câmera',
					message: 'Precisamos da sua permissão para usar a câmera',
					buttonPositive: 'Ok',
					buttonNegative: 'Cancelar',
				}}
			/>
		</SafeAreaView>
	)
}`

cenaHara, 2020-11-24:
+1 hi @callmephilip  @Luan-Sens @herokuapp88  any solution ?