react-native-camera/react-native-camera #2374: How to check if NO barcode is detected?

Esshahn, 2019-07-15:
With 

````
onGoogleVisionBarcodesDetected={this.barcode_recognized}
````

I can check for detected barcodes, but how do I check if NO barcode is visible? The issue is that when I remove a barcode from the camera view, the barcode is not removed from the barcodes array (see example files provided here). Therefore the overlay box that displays the barcode info keeps being displayed.

Thanks for any support!

saeid85, 2019-07-17:
Just a trick:
Add a timer, so if no barcode is detected for example in 30 seconds stop camera!

Esshahn, 2019-07-17:
Thanks @saeid85, I considered that but decided it would be too hacky.

closed 2019-09-22 by stale[bot]

Ashafix, 2020-03-18:
@Esshahn : Did you find a non-hacky solution?