react-native-camera/react-native-camera #2375: Error: Source file at path `file:///data/user/0/com.testing/cache/Camera/b05a1fae-5fd1-4559-9f50-9e44c2d4ba0d.jpg` does not exist

herokuapp88, 2019-07-15:
# Bug Report

react-native-camera version:

"git+https://git@github.com/react-native-community/react-native-camera.git"

**Platforms**

Android

**Versions**

react-native info
info 
  React Native Environment Info:
    System:
      OS: macOS 10.14.5
      CPU: (8) x64 Intel(R) Core(TM) i7-7820HQ CPU @ 2.90GHz
      Memory: 65.98 MB / 16.00 GB
      Shell: 3.2.57 - /bin/bash
    Binaries:
      Node: 11.4.0 - /usr/local/bin/node
      npm: 6.4.1 - /usr/local/bin/npm
      Watchman: 4.9.0 - /usr/local/bin/watchman
    SDKs:
      iOS SDK:
        Platforms: iOS 12.2, macOS 10.14, tvOS 12.2, watchOS 5.2
      Android SDK:
        API Levels: 23, 25, 26, 27, 28
        Build Tools: 27.0.3, 28.0.2, 28.0.3
        System Images: android-22 | Google APIs Intel x86 Atom, android-27 | Google APIs Intel x86 Atom, android-27 | Google Play Intel x86 Atom, android-28 | Google APIs Intel x86 Atom, android-28 | Google Play Intel x86 Atom
    IDEs:
      Android Studio: 3.1 AI-173.4907809
      Xcode: 10.2.1/10E1001 - /usr/bin/xcodebuild
    npmPackages:
      react: ^16.8.6 => 16.8.6 
      react-native: ^0.59.8 => 0.59.8 
    npmGlobalPackages:
      react-native-cli: 2.0.1
      react-native-cn-richtext-editor: 1.0.33
      react-native-git-upgrade: 0.2.7

**Description/Current Behaviour**
I am trying to use the following code to take picture and save to custom location on android device, but it threw exception:
Error: Source file at path `file:///data/user/0/com.testing/cache/Camera/b05a1fae-5fd1-4559-9f50-9e44c2d4ba0d.jpg` does not exist
The weird thing is I already check if the file exists, and it returns true, and when I do mv, then it says the file does not exist.

`try {
                    console.log("takePicture RNFetchBlob.fs.dirs.PictureDir: ", RNFetchBlob.fs.dirs.PictureDir);
                    const data = await this.camera.takePictureAsync();
                    console.log("data: ", data);
                    let exist = await RNFetchBlob.fs.exists(`${RNFetchBlob.fs.dirs.DCIMDir}/myfolder`);
                    console.log("exist: ", exist);
                    if(!exist)
                       await RNFetchBlob.fs.mkdir(`${RNFetchBlob.fs.dirs.DCIMDir}/myfolder`);
                    var sec = new Date().getSeconds(); 
                    console.log("sec: ", sec);
                    exist = await RNFetchBlob.fs.exists(`${data.uri}`);
                    console.log("takePicture exist: ", exist, data.uri);
                    if(exist){
                        await RNFetchBlob.fs.mv(
                            data.uri, 
                            `${RNFetchBlob.fs.dirs.DCIMDir}/myfolder/${sec}.jpg`
                        );
                        
                    }
                } catch (e) {
                    console.error("takePicture e: ", e);
                }`

**Expected Behaviour**
Expecting the file is moved to xx/myfolder directory

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

snake-eye77, 2019-07-16:
I have same problem

MateusAndrade, 2019-07-16:
@herokuapp88  you can test if you can access this file from your app? whiteout moving it? Giving a overview it looks related to RNFetchBlob or something similar to it.

lynrose10, 2019-07-29:
I also have a problem related to source file path
I'm using the path to upload in cloudinary but it always throws error "Invalid URL for upload"

Example from simulator
`file:///Users/<name>/Library/Developer/CoreSimulator/Devices/A283C891-3877-4491-B6F0-8CE988B7EAD6/data/Containers/Data/Application/FC93EDED-61B8-4571-952D-4A330187C9B1/Library/Caches/Camera/EF595B94-307B-4BE4-8215-4B666498A862.jpg`

[image]

similar error with IOS Simulator and Device

I also tried RNFetchBlob.fs.exists and returns `false`

SimonErm, 2019-09-06:
Try `data.uri.replace("file://","")` before moving then it should work. That's caused by an unconsistent behaviour of RNFetchBlob since they normalize the path before calling exist natively(https://github.com/joltup/rn-fetch-blob/blob/085db09bc9409a74143350b93341fd0f78d3eddc/android/src/main/java/com/RNFetchBlob/RNFetchBlobFS.java#L652) but not before moving the file.

HappyCMD, 2019-10-02:
@SimonErm Did not work for me,

I'm using data.url.split('///').pop() and rn-fetch-blob to move the file which gives me the correct path, but Source File not found.. it does display however when I do:

                    <Image 
                        source={{uri: data.uri}}
                        style={styles.imageWrapper}
                    />

But the cached file itself seems to not exist at that path

I tried with data.url.split('//').pop() too

SreevatsaAcharya, 2019-10-24:
hi, I am facing the same issue, any suggestions on how I can upload the image using URI? It used to work before but now when i upload the file  it does not seem to exist in the cache.

SimonErm, 2019-10-24:
Can you try with https://github.com/itinance/react-native-fs instead?

SreevatsaAcharya, 2019-10-24:
I am also using the document picker library and  images I pick from my internal storage upload fine.(url encoded image link.) but am unable to upload cached images of type jpeg. Definite that backend works fine.

closed 2019-12-30 by stale[bot]

aftabaminzoobiapps, 2020-10-24:
[Error: Source file at path `file:///storage/emulated/0/Download/VideoDownloader/TESTING_FOLDER/FB_VID_1603523704838.mp4 does not exist] some Error
`
this file exists on my phone. I have triple checked it. 

if it tried a path like this 
`/storage/emulated/0/Download/VideoDownloader/TESTING_FOLDER/FB_VID_1603523704838.mp4`
then this error

[Error: Source file not found.] some Error

Here is my source code 

` const filemove = async () =>{
       await RNFetchBlob.fs.mv( '/storage/emulated/0/Download/VideoDownloader/TESTING_FOLDER/FB_VID_1603523704838.mp4',
  `${RNFetchBlob.fs.dirs.SDCardDir}/VideoDownloader/TESTING_FOLDER/`
          ).then(() => {
              console.log('File Downloaded')
          }).catch((err) =>{
            console.log(err,'some Error')
          })
      }`