react-native-camera/react-native-camera #2376: App Crashing everytime

jothikannan, 2019-07-15:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
Yes
- [ ] Did you try master?
Yes
- [ ] Did you look for existing matching issues?
N
**Platforms**
Android

**Versions**

- Android: 9
- react-native-camera: ^2.11.1
- react-native: 0.60.3
- react: 16.8.6
  
^3.11.1
**Description/Current Behaviour**
App Crashing

After adding `missingDimensionStrategy 'react-native-camera', 'general'` in android/app/build.gradle my application is crashing
**Expected Behaviour**
It should work without crashing the app
I have tried with multiple instances with the same versions of all the module, before installing all 
the components app was working well when I install react-native-camera my app getting crash on react-native run-android

OscarXal, 2019-07-15:
I just facing that problem today, but I had:
"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera.git"

 I tried with version 2.11.0 and all works fine.

devepro, 2019-07-16:
@jothikannan i am also facing the same issue

@OscarXal i tried using 2.11.0 but still facing the same issue. Is your project migrated to androidx? Mine is not. So could that cause this issue?

jothikannan, 2019-07-16:
@OscarXal My project is androidx migrated so I believe something wrong with the react-native-camera. someone, please fix it, or let us know what's wrong.

jothikannan, 2019-07-16:
I figured out that the react-native-camera is not working with `react-native: 0.60.3` i have downgraded into `react-native: 0.60.0` now my app is working without any crash.

closed 2019-07-16 by jothikannan