react-native-camera/react-native-camera #2379: Ability to select back camera for multi-camera devices

Pauly-D, 2019-07-17:
# Feature-Request

**Description**
Ability to select which **back** camera to use. Particularly useful for all the multi-camera android devices with wide angle lenses. 

**Possible Implementations**
Since the implementation of [Android API Level 9](https://developer.android.com/training/camera/cameradirect#java) the Camera object supports the passing of an id to select which camera to use (eg. the wide angel lens). 

Which files need to be modified to take advantage of the new android API so that we can update the **type={RNCamera.Constants.Type.back}** for selection of different cameras on the back of android devices?

cristianoccazinsp, 2019-09-17:
Hey, any luck with this?

cristianoccazinsp, 2019-11-16:
I believe this is now possible with the above PRs and can be closed.

closed 2019-11-16 by sibelius

farhanmeo, 2021-06-07:
Unable to use Wide angle camera in android

cristianoccazinsp, 2021-06-07:
Sadly, newer android devices no longer support camera enumeration through the currently implemented method. For these devices, a new implementation path must be followed in order to be able to read from individual logical cameras.

I believe the Camera2 implementation may be able to support that, but it is way too bugged to use right now.