react-native-camera/react-native-camera #2380: Reduce the video recording size

Suhas404, 2019-07-18:
I'm recording the video, but the size of the recording for 3 mins is very huge. and size of the video recording varies from device to device. but they in huge size in the end compared to native android or ios app. currently, I'm using the least quality available as per the doc, i.e  `RNCamera.Constants.VideoQuality["4:3"]`  for both Android and ios. is there any way we can reduce the size of the video or any video compression methods available. 

`startRecording = async function() {
    clearInterval(timer_handler1);
    this.setState({ recording: true });
    if (this.camera) {
      this.camera
        .recordAsync({
          quality: RNCamera.Constants.VideoQuality["4:3"],
          forceUpOrientation: true,
          fixOrientation: true,
          //
          quality: 1,
          videoOrientation: 1,
          deviceOrientation: 1,
        })
        .then(data => {
          this.setState(
            {
              FilePath: data
            },
            () => {
              this.saveVideo(data["uri"]);
            }
          );
        })
        .catch(error => {
          console.log(error);
        });
    }
  };`

adamsolomon1986, 2020-04-05:
I don't understand why this issue doesn't have any activity. I'm stuggling with android now. Recording 30 seconds on Google Pixel 3 with ['16:9'] ratio and ['480p'] quality results consistently in ~21MB File......... any ideas?

Suhas404, 2020-04-08:
I'm waiting so long for someone from the community member to reply. Stuck with the same issue here!

sibelius, 2020-04-08:
this pull request could help https://github.com/react-native-community/react-native-camera/pull/2766

it enables grab the value of preview so you can adjust properly

fabriziobertoglio1987, 2020-04-08:
iOS compresses the video, android does not. You need to use utility to compress the video and the send it to your server. I suggest you to use utilities like `ffmpeg.js` to compress the video on client side.

https://github.com/shahen94/react-native-video-processing
https://github.com/Kagami/ffmpeg.js/

mlazari, 2020-05-06:
Did you try `videoBitrate` [setting](https://react-native-community.github.io/react-native-camera/docs/rncamera#recordasync-options-promise)? I see it's available on both iOS and Android. It should give you control on file size, though lower value results in lower quality.

alkalox, 2020-05-26:
Yes, use the `videoBitrate` option. I have it set to `0.25*1000*1000`. 2 min of Video results in ~5MB file size with 480p quality.

closed 2020-08-01 by stale[bot]