react-native-camera/react-native-camera #2382: FaCC faster than pendingAuthorizationView/notAuthorizedView + children?

cristianocca, 2019-07-20:
# Bug Report

**To Do First**

- [ ] Did you try latest release? Yes
- [ ] Did you try master? No
- [ ] Did you look for existing matching issues? Yes

**Platforms**
Android

**Versions**

- Android: 8.1 and 9
- react-native-camera: 2.11.0
- react-native: 0.59.9
- react: 16.8.6

**Description/Current Behaviour**
Following issue https://github.com/react-native-community/react-native-camera/issues/1768 I'm facing something odd, which I guess no one has reported before or even noticed because no one really uses FaCC (otherwise that issue would have been fixed by now).

Basically, when using children instead if FaCC, I am now noticing that the pending/loading indicator I was previously using shows for up to 2 seconds, as opposed to being nearly instant when using FaCC. 

Has anyone seen this? Why is the FaCC so different?

cristianocca, 2019-07-20:
My only guess is that the FaCC doesn't wait for permissions or something (reason it crashes on IOS). I've tested pretty much everything to make sure it wasn't related to code on my side, but not using the FaCC option makes the camera load quite slow!

closed 2019-09-25 by stale[bot]

cristianoccazinsp, 2019-10-27:
This can be closed with https://github.com/react-native-community/react-native-camera/pull/2560

I believe that the visible slowdown was related to the heavy UI work and camera starting at the same time, and the redundant setup/start calls that were happening and somehow avoided when using FACC.