react-native-camera/react-native-camera #2383: /node_modules/react-native-camera/ios/RN/RNFaceDetectorModuleMLKit.h:1:9: 'React/RCTBridgeModule.h' file not found

renalpraba, 2019-07-21:
# Bug Report

**To Do First**

- [v] Did you try latest release?
- [v] Did you try master?
- [v] Did you look for existing matching issues?

**Platforms**

On Android Works
Not On iOS

**Versions**

- react-native-camera: 3.0.0
- react-native: 0.60.4
- react: 16.8.6
- react-navigation: 3.11.1

**Description/Current Behaviour**

Give me error

**Expected Behaviour**

Build Successful

**Steps to Reproduce**

Install & Autolink then run

**Additionals**

[image]
[image]

renalpraba, 2019-07-21:
I am using react-native 0.60.x default podfile setting instead the suggestion on installation.md
my podfile is like this

```
platform :ios, '9.0'
require_relative '../node_modules/@react-native-community/cli-platform-ios/native_modules'

target 'Template' do
  # Pods for Template
  pod 'React', :path => '../node_modules/react-native/'
  pod 'React-Core', :path => '../node_modules/react-native/React'
  pod 'React-DevSupport', :path => '../node_modules/react-native/React'
  pod 'React-fishhook', :path => '../node_modules/react-native/Libraries/fishhook'
  pod 'React-RCTActionSheet', :path => '../node_modules/react-native/Libraries/ActionSheetIOS'
  pod 'React-RCTAnimation', :path => '../node_modules/react-native/Libraries/NativeAnimation'
  pod 'React-RCTBlob', :path => '../node_modules/react-native/Libraries/Blob'
  pod 'React-RCTImage', :path => '../node_modules/react-native/Libraries/Image'
  pod 'React-RCTLinking', :path => '../node_modules/react-native/Libraries/LinkingIOS'
  pod 'React-RCTNetwork', :path => '../node_modules/react-native/Libraries/Network'
  pod 'React-RCTSettings', :path => '../node_modules/react-native/Libraries/Settings'
  pod 'React-RCTText', :path => '../node_modules/react-native/Libraries/Text'
  pod 'React-RCTVibration', :path => '../node_modules/react-native/Libraries/Vibration'
  pod 'React-RCTWebSocket', :path => '../node_modules/react-native/Libraries/WebSocket'

  pod 'React-cxxreact', :path => '../node_modules/react-native/ReactCommon/cxxreact'
  pod 'React-jsi', :path => '../node_modules/react-native/ReactCommon/jsi'
  pod 'React-jsiexecutor', :path => '../node_modules/react-native/ReactCommon/jsiexecutor'
  pod 'React-jsinspector', :path => '../node_modules/react-native/ReactCommon/jsinspector'
  pod 'yoga', :path => '../node_modules/react-native/ReactCommon/yoga'

  pod 'DoubleConversion', :podspec => '../node_modules/react-native/third-party-podspecs/DoubleConversion.podspec'
  pod 'glog', :podspec => '../node_modules/react-native/third-party-podspecs/glog.podspec'
  pod 'Folly', :podspec => '../node_modules/react-native/third-party-podspecs/Folly.podspec'
  
  # pod 'RNFirebase', :path => '../node_modules/react-native-firebase'
  pod 'GoogleMaps'
  pod 'Google-Maps-iOS-Utils'
  pod 'Firebase/Core', '~> 6.3.0'
  pod 'Firebase/Messaging', '~> 6.3.0'

  pod 'react-native-camera', path: '../node_modules/react-native-camera'

  use_native_modules!

end

post_install do |installer|
  installer.pods_project.targets.each do |target|
    if target.name == 'react-native-google-maps'
      target.build_configurations.each do |config|
        config.build_settings['CLANG_ENABLE_MODULES'] = 'No'
      end
    end
    if target.name == "React"
      target.remove_from_project
    end
  end
end

```

renalpraba, 2019-07-21:
_Note: I am actually not using Face Detection feature on my project_

renalpraba, 2019-07-21:
I change the import from <React/RCT...> to #import "RCT...",
the file is found, but there is a lot of file need to change import like that after

renalpraba, 2019-07-21:
I think it's my fault,
previously I got nativemodules.requestpermission error then I thought I must manual link it
after adding xcodeproj to libraries this error shows

I think on installation.md should add instruction to install react-native-permission package, I facing this error because no information about it

closed 2019-07-24 by renalpraba

danielcsrs, 2019-11-20:
If still useful. I also had a similar problem, but in this case a have two problems.

1. The reference to MLKit is missing in podfile.
2. Probably the reference for RNCamera is't configured correctly into Xcode.

For two problems, [check the docs for installation here,](https://github.com/react-native-community/react-native-camera/blob/master/docs/installation.md) this explain how to configure react-native-camera with FaceDetector.

Well, in my case, RCTBridgeFile i was missing when starting build. The solution was make:

1. Clean the builds. On the Xcode opened, navigate to Product > Clean Build Folder.
2. Into project on VSCode, open the podfile in ios folder, modify the dependency as per the docs.
3. Into project on VSCode, open the termnal, navigate to ios folder (_run cd ios_) and run command pod install.
4. Open [YourAppName].xcworskpace into Xcode.
5. Make manual installation, [as indicated in the documentation](https://github.com/react-native-community/react-native-camera/blob/master/docs/installation.md#ios).
6. Buil your app.

I hope it helps. :)