react-native-camera/react-native-camera #2384: Task :app:processDebugManifest FAILED

DerekFei, 2019-07-22:
react-native-camera: version: 3.0.0 
upgrade from 2.11.1

after react-native run-android, this error occurs 

> Task :react-native-camera:compileMlkitDebugJavaWithJavac
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: C:\*project-dir*\node_modules\react-native-camera\android\src\main\java\com\google\android\cameraview\Camera2.java uses unchecked or unsafe 
operations.
Note: Recompile with -Xlint:unchecked for details.

> Task :app:processDebugManifest FAILED
projectdir\android\app\src\debug\AndroidManifest.xml:22:18-86 Error:
        Attribute application@appComponentFactory value=(androidx.core.app.CoreComponentFactory) from [androidx.core:core:1.0.0] AndroidManifest.xml:22:18-86
        is also present at [com.android.support:support-compat:28.0.0] AndroidManifest.xml:22:18-91 value=(android.support.v4.app.CoreComponentFactory).
        Suggestion: add 'tools:replace="android:appComponentFactory"' to <application> element at AndroidManifest.xml:7:5-117 to override.

See http://g.co/androidstudio/manifest-merger for more information about the manifest merger.

FAILURE: Build failed with an exception.

spwuc, 2019-07-23:
This problem has been bothering me for a long time. It took a day to find out that it was because of the library's problems. 3.0.0 introduced androidx. Support and Androidx are in conflict, so download 2.11.1

hntddt123, 2019-07-25:
https://github.com/react-native-community/react-native-camera/issues/2320#issuecomment-503628961

rajpootathar, 2019-08-18:
I am currently facing the same issue

fabriziobertoglio1987, 2019-08-22:
I am currently facing this issue, It is connected to `react-native-camera`.

The issue stopped after commenting the `react-native-camera` lines inside `settings.gradle`, `MainApplication.java` and `build.gradle`

MuhammadUsman786786, 2019-08-23:
@fabriziobertoglio1987  solution works for me

closed 2019-10-29 by stale[bot]