react-native-camera/react-native-camera #2388: Camera preview fixed height increase when camera display.

SA-KBJ, 2019-07-24:
I have set RNCamera height to fixed 500. When the camera was not loaded is height is 500. But while preview show than there height increase.

I found the same behavior in library basic example: https://github.com/react-native-community/react-native-camera/tree/master/examples/basic.

Style
```
container: {
        flex: 1,
        flexDirection: 'column',
        backgroundColor: colors.profileTabSelectedColor
    },
    preview: {
        flexDirection: 'column',
        alignItems: 'center',
        width: dimen.value_100wp,
        height: 500,
        backgroundColor: colors.profileTabSelectedColor
    }
}
```
Camera rendering
```
 <RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          style={createPostStyle.preview}
          type={RNCamera.Constants.Type.back}
          flashMode={this.state.flash}
          androidCameraPermissionOptions={{
            title: string.alert.CameraPÒermissionTitle,
            message: string.alert.CameraPermissionNote,
            buttonPositive: string.alert.buttonOk,
            buttonNegative: string.alert.buttonCancel,
          }}
        />
```
ScreenShoot
[image]

[image]

kelokchan, 2019-08-10:
Try to set `overflow: 'hidden'` to your container

SA-KBJ, 2019-08-10:
@kelokchan I have tried to set **overflow: 'hidden'** to the container but still same behavior happens.

Obi1Kennoby, 2019-10-08:
Add `overflow: 'hidden`' not to container but to your `preview` style. it helped to me

dca-ps, 2019-10-08:
I'm also with this problem, but @Obi1Kennoby tip helped a lot!

closed 2019-12-14 by stale[bot]

sarannandhakumar, 2020-02-26:
> Try to set `overflow: 'hidden'` to your container
setting overflow : 'hidden' works fine, but the hidden portion(hide using overflow:'hidden') of camera view also reflecting in image after capture. Is there any way to stretch the camera view ?.

Sargnec, 2020-08-17:
> > Try to set `overflow: 'hidden'` to your container
> > setting overflow : 'hidden' works fine, but the hidden portion(hide using overflow:'hidden') of camera view also reflecting in image after capture. Is there any way to stretch the camera view ?.

Have this same issue. Hidden part is visible after taken picture

realabbas, 2022-09-20:
> overflow: 'hidden'

Yeah works for me! Thanks