react-native-camera/react-native-camera #2390: Does barcode reading require Google Firebase? Privacy issues?

aschrijver, 2019-07-25:
# Question

**To Do First**
- [x] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

From the documentation it is not totally clear to me whether any barcode reading application using `react-native-camera` automatically means that Google Firebase is being used under the covers. I just added `v2.11.1` to a react-native app that will be used for scanning supermarket purchases, and privacy means that product scans should not be sent to Google.

Can I use this version of `react-native-camera` for barcode scanning in a way where privacy is guaranteed, and if not, are there older versions that do not require the online connection to Google?

In any case, as a member of the [Humane Tech Community](https://humanetech.community) I think the README and website should clearly document the dependencies on Google Firebase, so other devs are aware of privacy concerns.

**Tags**

- privacy
- google
- firebase
- humanetech

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

aschrijver, 2019-07-25:
My current understanding is that `onBarCodeRead` uses local code and `onGoogleVisionBarcodesDetected` uses Firebase.

andresmtz98, 2019-08-06:
> My current understanding is that `onBarCodeRead` uses local code and `onGoogleVisionBarcodesDetected` uses Firebase.

Yes, both don't require internet connection. I'm using the Firebase MLKit for barcode detection.

aschrijver, 2019-08-07:
Thanks for the info!

closed 2019-08-07 by aschrijver

MakhouT, 2020-01-06:
I also was looking for this, the docs weren't really clear about it though.
Is there an advantage of using the firebase one instead of the local one?