react-native-camera/react-native-camera #2391: does not work with react-native 0.60.4 on android

dmitryame, 2019-07-26:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:
  

**Description/Current Behaviour**

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

dmitryame, 2019-07-26:
Migrating existing project to react-native 0.60.4, fails to run on android and fails on react-native-camera
`BUILD FAILED in 1s

error Failed to install the app. Make sure you have the Android development environment set up: https://facebook.github.io/react-native/docs/getting-started.html#android-development-environment. Run CLI with --verbose flag for more details.
Error: Command failed: ./gradlew app:installDebug -PreactNativeDevServerPort=8081

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:preDebugBuild'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve project :react-native-camera.
     Required by:
         project :app
      > Cannot choose between the following variants of project :react-native-camera:
          - generalDebugRuntimeElements
          - mlkitDebugRuntimeElements
        All of them match the consumer attributes:
          - Variant 'generalDebugRuntimeElements' capability AwesomeProject:react-native-camera:unspecified:
              - Unmatched attributes:
                  - Found com.android.build.api.attributes.VariantAttr 'generalDebug' but wasn't required.
                  - Found react-native-camera 'general' but wasn't required.
              - Compatible attributes:
                  - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
                  - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
                  - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.
          - Variant 'mlkitDebugRuntimeElements' capability AwesomeProject:react-native-camera:unspecified:
              - Unmatched attributes:
                  - Found com.android.build.api.attributes.VariantAttr 'mlkitDebug' but wasn't required.
                  - Found react-native-camera 'mlkit' but wasn't required.
              - Compatible attributes:
                  - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
                  - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
                  - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.
`

dmitryame, 2019-07-26:
Just to make sure I'm not doing anything crazy in my project, I tried to 

`react-native init AwesomeProject`

As soon as I add react-native-camera to the dependencies list `yarn add react-native-camera` , I can not longer launch the project on android `react-native run-android` -- it fails with exactly the same error (see my previous comment).

sibelius, 2019-07-26:
have you read the docs https://react-native-community.github.io/react-native-camera/docs/installation.html#android?

you are missing `missingDimensionStrategy`

closed 2019-07-26 by sibelius

dmitryame, 2019-07-26:
It definitely solved my issue, thank you. 
I wish there were more explicit references in the doc to 0.60.x new autolinking process. It's just not always obvious which steps apply to 0.59 and which to 0.60. 
Thank you.

sibelius, 2019-07-26:
feel free to send a PR improving it

talaikis, 2019-07-27:
With empty project (react-native 0.60.4, react-native-camera 3.0.0), and manual installation (because autolinking doesn't work properly) we're getting "Native module RCTCameraModule tried to override RCTCameraModule."

**The solution** is not to include ```new RNCameraPackage()``` into  ```getPackages()``` and everything is starting to work fine.

RyanThomas73, 2019-08-07:
@talaikis the error you're describing sounds like you are both autolinking and manually linking causing the package to be added twice.

If you're manually linking then disable autolinking by adding an entry in `react-native.config.js`

```
module.exports = {
    .... other stuff ...
    dependencies: {
        "react-native-camera": {
            platforms: {
                android: null // disable autolinking on android because it doesn't work correctly right now
            }
        }
    }
}
```

https://github.com/react-native-community/cli/blob/master/docs/autolinking.md#how-can-i-disable-autolinking-for-unsupported-library

DenisDov, 2019-09-11:
Manual linking not required. Simply add `missingDimensionStrategy 'react-native-camera', 'general'` inside `defaultConfig` in `android/app/build.gradle`

brendonhc, 2019-11-11:
> Manual linking not required. Simply add `missingDimensionStrategy 'react-native-camera', 'general'` inside `defaultConfig` in `android/app/build.gradle`

It's perfect! Worked fine for me!!
I'm using react-native: 0.60.5, just do it:

`yarn add react-native-camera`

In the file `android/app/build.gradle` add
```
defaultConfig {
    // ... block of the default code...
    missingDimensionStrategy 'react-native-camera', 'general'
}
```

Finally, run `react-native run-android` and be happy!

Thanks! 👍

klodha, 2020-03-23:
Trying to understand why this extra step is required? Adding missingDimensionStrategy worked, but does that means some auto configuration failing or missing? As with other packages, usually this step is not needed.