react-native-camera/react-native-camera #2394: checkIfRecordAudioPermissionsAreDefined

idoor88, 2019-07-27:
# Bug Report
I got the most recent 3.0.0 version of the library, but when I tried to capture a camera, I got this exception:
Unhandled promise rejection TypeError: CameraManager.checkIfRecordAudioPermissionsAreDefined is not a function
    at requestPermissions$ (RNCamera.js:54)
    at tryCatch (runtime.js:45)
    at Generator.invoke [as _invoke] (runtime.js:271)
    at Generator.prototype.<computed> [as next] (runtime.js:97)
    at tryCatch (runtime.js:45)
    at invoke (runtime.js:135)
    at runtime.js:145
    at run (es6.promise.js:75)
    at es6.promise.js:92
    at flush (_microtask.js:18)

**Platforms**
My testing platform is Android

and installed with this command:

npm install --save react-native-camera@git+https://git@github.com/react-native-community/react-native-camera.git
react-native link react-native-camera

**Expected Behaviour**

No exception and live camera view should show up

**Steps to Reproduce**

Use the following code:

`<View style={styles.preview}>
                    <RNCamera
                        ref={(ref) => {
                            this.camera = ref;
                        }}
                        style={styles.camera}   //preview2
                        type={this.state.type}
                        flashMode={this.state.flashMode}
                        permissionDialogTitle="Permission to use camera"
                        permissionDialogMessage="We need your permission to use your camera phone"
                        ratio="1:1" // "1:1" reload app will not show 1:1 not supported issue
                        androidCameraPermissionOptions={{
                            title: 'Permission to use camera',
                            message: 'We need your permission to use your camera',
                            buttonPositive: 'Ok',
                            buttonNegative: 'Cancel',
                          }}
                          androidRecordAudioPermissionOptions={{
                            title: 'Permission to use audio recording',
                            message: 'We need your permission to use your audio',
                            buttonPositive: 'Ok',
                            buttonNegative: 'Cancel',
                          }}
                          onGoogleVisionBarcodesDetected={({ barcodes }) => {
                            console.log(barcodes);
                          }}
                    />
                </View>`

also followed instructions to set the permissions:
Permissions
To use the camera,

On Android you must ask for camera permission:
  <uses-permission android:name="android.permission.CAMERA" />
To enable video recording feature you have to add the following code to the AndroidManifest.xml:

  <uses-permission android:name="android.permission.RECORD_AUDIO"/>
  <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
  <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE" />

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

My environment after running react-native info:
info Fetching system and libraries information...
System:
    OS: macOS 10.14.5
    CPU: (8) x64 Intel(R) Core(TM) i7-7820HQ CPU @ 2.90GHz
    Memory: 366.11 MB / 16.00 GB
    Shell: 3.2.57 - /bin/bash
  Binaries:
    Node: 11.4.0 - /usr/local/bin/node
    npm: 6.4.1 - /usr/local/bin/npm
    Watchman: 4.9.0 - /usr/local/bin/watchman
  SDKs:
    iOS SDK:
      Platforms: iOS 12.4, macOS 10.14, tvOS 12.4, watchOS 5.3
    Android SDK:
      API Levels: 23, 25, 26, 27, 28
      Build Tools: 27.0.3, 28.0.2, 28.0.3
      System Images: android-22 | Google APIs Intel x86 Atom, android-27 | Google APIs Intel x86 Atom, android-27 | Google Play Intel x86 Atom, android-28 | Google APIs Intel x86 Atom, android-28 | Google Play Intel x86 Atom
  IDEs:
    Android Studio: 3.1 AI-173.4907809
    Xcode: 10.3/10G8 - /usr/bin/xcodebuild
  npmPackages:
    react: ^16.8.6 => 16.8.6 
    react-native: ^0.60.0 => 0.60.0 
  npmGlobalPackages:
    react-native-cli: 2.0.1
    react-native-cn-richtext-editor: 1.0.33
    react-native-git-upgrade: 0.2.7

idoor88, 2019-07-27:
unlink and uninstall, then install, link again

closed 2019-07-27 by idoor88

IgorDePaula, 2019-12-28:
I receiverd the same error in android, with the rights permissions.

sirpy, 2020-01-15:
I get the same error
if i set captureAudio={false} i get invariant violation exception about RNCamera

sirpy, 2020-01-15:
doesnt happen with master branch

shui2464, 2020-06-01:
so am i, it resolved ?

AGDVS1, 2020-11-03:
I have the same issue and 2020, my conclusion this library sucks!

kaushalkoladiya, 2021-01-08:
Gatting same issue from past few days and can't able to solve.
Please give some helpful instruction/resource to the outcome of this issue.

MateusAndrade, 2021-01-08:
> Gatting same issue from past few days and can't able to solve.
> Please give some helpful instruction/resource to the outcome of this issue.

Which version are you using? Could you please give it a try with [v3.14.0](https://github.com/react-native-camera/react-native-camera/releases/tag/v3.41.0)

kaushalkoladiya, 2021-01-13:
> > Gatting same issue from past few days and can't able to solve.
> > Please give some helpful instruction/resource to the outcome of this issue.
> 
> Which version are you using? Could you please give it a try with [v3.14.0](https://github.com/react-native-camera/react-native-camera/releases/tag/v3.41.0)

I have also tried with v3.14.0
Sorry to say, but, nothing new happened same error.

Andriiklymiuk, 2021-09-03:
have anyone found the solution for this one?