react-native-camera/react-native-camera #2395: Face detection working in iphone 6s, not working on Iphone X

TranHanHieu, 2019-07-30:
# Bug Report

**To Do First**

- [ ] Did you try latest release?

**Platforms**

**Versions**

- ios: 12
- react-native-camera:
- react-native: 0.60.0
- react: 16.8.6
  

**Description/Current Behaviour**
iPhone 6S working (iOS 12.1.2), iphone X is currently not detecting any faces. On a component mount, facedetection callback called once with empty

{ 
  faces: []
  type: "face"
}

**Expected Behaviour**

Face detection should work on iPhone 6S

**Steps to Reproduce**

Install the package using CocoaPods and run on a device, try to detect faces.

Esxiel, 2019-07-31:
Let me add to this.
I tried latest version so far (3.0.0), along with the latest version of react-native (0.60.4)
It works on my iPhone 6(12.2), but doesn't work on my iPhone Xs 12.3.1

Esxiel, 2019-07-31:
I've found out that the face detection on the iPhoneX works if we switch the phone orientation to landscape mode.

TranHanHieu, 2019-07-31:
Thanks @Esxiel, but I think the problem is not orientation to landspace, because the day before it worked with the old build on iphone X but didn't work on 6s. I think there are issues in some issues. I need help

TranHanHieu, 2019-08-14:
FaceDetectorManagerMlkit: 

- (void)findFacesInFrame:(UIImage *)uiImage
                  scaleX:(float)scaleX
                  scaleY:(float)scaleY
               completed:(void (^)(NSArray *result))completed 
{
    self.scaleX = scaleX;
    self.scaleY = scaleY;

    FIRVisionImage *image = [[FIRVisionImage alloc] initWithImage:uiImage];
    NSMutableArray *emptyResult = [[NSMutableArray alloc] init];
    [_faceRecognizer
     processImage:image
     completion:^(NSArray<FIRVisionFace *> *faces, NSError *error) {
        RCTLog(@"faces: %@", faces);
        RCTLog(@"error: %@", error);

         if (error != nil || faces == nil) {
             completed(emptyResult);
         } else {
             completed([self processFaces:faces]);
         }
     }];
}

On iphone x log faces, error is nil.

votanhung, 2019-08-23:
> I've found out that the face detection on the iPhoneX works if we switch the phone orientation to landscape mode.
> 
> Maybe #2221 needs to be reopened?

iPhone XS MAX has the same. It only works on landscape mode.

kiritnim, 2019-09-10:
> > I've found out that the face detection on the iPhoneX works if we switch the phone orientation to landscape mode.
> > Maybe #2221 needs to be reopened?
> 
> iPhone XS MAX has the same. It only works on landscape mode.

@votanhung Yes it's working in landscape, you have solution of portrait.

GH974, 2019-09-16:
Hi, i tested on the last version and it work only in landscape mode on iphone X

closed 2019-11-22 by stale[bot]

sushant1333, 2019-11-26:
Hi, I am still facing the issue.

Faces are detected only in landscape mode, but not in portrait mode.
Any solution to it?

camchis, 2021-07-01:
This seems to still be an issue for me... turning to landscape mode isn't a suitable solution