react-native-camera/react-native-camera #2411: Android camera doesn't open in type FRONT

iagorm, 2019-08-04:
# Bug Report
I'm facing an issue which is, my camera does not open when the type is `front`, in `back` it shows up.
**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**
RN 59.9

Issue happens on Android.

**Versions**
RN-camera: 2.11.1

- Android: Xiaomi Mi 8 lite - Android 9
- react-native-camera: 2.11.1
- react-native: 0.59.9
- react: 16.8.3
  

**Description/Current Behaviour**
The camera doesn't show up when the type is set to `front`.

**Expected Behaviour**
The camera must appear.

**Steps to Reproduce**
Set Camera's component to type `front`.

**Additionals**
trace:

```
8-04 17:11:01.108  7779  7779 E unknown:ReactNative: NullPointerException when executing ViewGroup.dispatchDraw method
08-04 17:11:01.108  7779  7779 E unknown:ReactNative: java.lang.NullPointerException: Attempt to invoke interface method 'java.lang.Object java.util.SortedSet.last()' on a null object reference
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at com.google.android.cameraview.Camera1.adjustCameraParameters(Camera1.java:700)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at com.google.android.cameraview.Camera1$1.onSurfaceChanged(Camera1.java:133)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at com.google.android.cameraview.PreviewImpl.dispatchSurfaceChanged(PreviewImpl.java:56)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at com.google.android.cameraview.TextureViewPreview$1.onSurfaceTextureAvailable(TextureViewPreview.java:46)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.TextureView.getTextureLayer(TextureView.java:390)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.TextureView.draw(TextureView.java:339)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.updateDisplayListIfDirty(View.java:19109)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.draw(View.java:19962)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.drawChild(ViewGroup.java:4337)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4114)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.draw(View.java:20237)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.updateDisplayListIfDirty(View.java:19109)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.draw(View.java:19962)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.drawChild(ViewGroup.java:4337)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.dispatchDraw(ViewGroup.java:4114)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at com.facebook.react.views.view.ReactViewGroup.dispatchDraw(ReactViewGroup.java:684)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.draw(View.java:20237)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.updateDisplayListIfDirty(View.java:19109)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4321)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4294)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.updateDisplayListIfDirty(View.java:19069)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4321)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4294)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.updateDisplayListIfDirty(View.java:19069)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4321)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4294)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.updateDisplayListIfDirty(View.java:19069)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4321)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4294)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.updateDisplayListIfDirty(View.java:19069)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4321)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4294)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.updateDisplayListIfDirty(View.java:19069)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4321)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4294)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.updateDisplayListIfDirty(View.java:19069)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4321)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4294)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.updateDisplayListIfDirty(View.java:19069)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4321)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4294)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.updateDisplayListIfDirty(View.java:19069)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4321)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4294)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.updateDisplayListIfDirty(View.java:19069)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4321)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4294)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.updateDisplayListIfDirty(View.java:19069)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4321)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4294)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.updateDisplayListIfDirty(View.java:19069)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4321)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4294)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.updateDisplayListIfDirty(View.java:19069)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4321)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4294)
08-04 17:11:01.108  7779  7779 E unknown:ReactNative:   at android.view.View.u
08-04 17:11:01.113  3095  3095 D KeyguardUpdateMonitor: received broadcast android.intent.action.BATTERY_CHANGED
```

MateusAndrade, 2019-08-05:
Did you tried with the master? This happens, when you start the component with type set to `FRONT` or whenever you change the camera type to `FRONT`?

iagorm, 2019-08-05:
@MateusAndrade, no I haven't tested master, but the master branch is AndroidX no? Also about the type is set always as `front`.

MateusAndrade, 2019-08-06:
@iagorm yes, but you can give a try at the example project applying this behavior. I will give a try at the example here, and see if i can reproduce your problem. :)

iagorm, 2019-08-06:
The issue is happening in a Xiaomi mi 8 Lite at the moment.

closed 2019-08-16 by iagorm