react-native-camera/react-native-camera #2412: Enable MLKit face detection while recording a video

tomo25, 2019-08-05:
# Question

**Ask your Question**
Hi, mlkit face detection is great but it stops working when I start recording a video.
 Is there a way to enable mlkit face detection while recording as well?

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

tomo25, 2019-08-12:
Does anyone have an idea what is causing this issue?

MateusAndrade, 2019-08-13:
Hi @tomo25 , i dont know if we got this feature on react-native-camera at the moment. I will try giving a look at this on this week. However, prs are always welcome! :smiley:

tomo25, 2019-08-15:
Thank you @MateusAndrade !

tomo25, 2019-08-27:
@MateusAndrade Hi! How is it going with this?

closed 2019-11-02 by stale[bot]