react-native-camera/react-native-camera #2413: RNCamera.checkVideoAuthorizationStatus is not a function

davebro, 2019-08-05:
# Question

**To Do First**
- [x] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
Migrating to v2.11.1 from a much older v0.10.0, I have gone through all the install steps in the README, our project has all the required keys in our Info.plist files. I have also manually linked the project iOS and the project builds successfully. Android runs just fine, iOS however throws an error at runtime when `RNCamera.checkVideoAuthorizationStatus` is called:
 
[image]

Has this function been deprecated in newer versions, and if so, is there a more preferred way to check for permissions on iOS? A quick search for `checkVideoAuthorizationStatus` in the react-native-camera repo shows the function is still part of the project in some capacity.

MateusAndrade, 2019-08-06:
Hi @davebro , i think this method has been deprecated or removed on the newer versions. You tried using ```refreshAuthorizationStatus()?```. Please, give a look at: https://react-native-community.github.io/react-native-camera/docs/api and let me know if this helps you.

takitus, 2019-08-06:
This is an issue on version 0.60.0 + of react-native. You can get around this for now by using a version lower than 60 (59.8 works). 

react-native init ProjName --version react-native@0.59.8

Ive obviously run into the same issue. Hopefully it gets rectified sooner rather than later.

MateusAndrade, 2019-08-06:
> Ive obviously run into the same issue. Hopefully it gets rectified sooner rather than later.

Thanks for letting us know that. 

@davebro which version of react-native are you using?

davebro, 2019-08-06:
@MateusAndrade Currently we are on react-native 0.58.6. We are upgrading shortly to 0.59.10 and I have also tested on that version, same result.

takitus, 2019-08-06:
if you are using an older version, just make sure you use 
"react-native link react-native-camera". 

It should resolve that issue

davebro, 2019-08-06:
In older versions of react-native-camera, I noticed whereas `checkVideoAuthorizationStatus` was a static function accessed like so:
```
import Camera from 'react-native-camera';

Camera.checkVideoAuthorizationStatus().then((isAuthorized) => {
    ...
})
```

The current version provides `refreshAuthorizationStatus` and requires an instance of RNCamera from `import { RNCamera } from 'react-native-camera';` in order to use it.

From the docs here https://react-native-community.github.io/react-native-camera/docs/api it seems like `refreshAuthorizationStatus` provides similar functionality? I'll have to test that, but if anyone has further insights, that'd be most helpful.

davebro, 2019-08-06:
@takitus Yes, `react-native link react-native-camera` has been run several times but still shows the same error. I've also manually linked the library, but with no success regarding `checkVideoAuthorizationStatus `.

takitus, 2019-08-06:
```
react-native init ProjName --version react-native@0.59.10
cd ProjName
yarn add react-native-camera
react-native link react-native-camera
```
*add permissions into info.plist*
```
<key>NSCameraUsageDescription</key>
<string>Your message to user when the camera is accessed for the first time</string>
<key>NSPhotoLibraryUsageDescription</key>
<string>Your message to user when the photo library is accessed for the first time</string>
<key>NSMicrophoneUsageDescription</key>
<string>Your message to user when the microphone is accessed for the first time</string>
<key>NSPhotoLibraryAddUsageDescription</key>
<string>Your message to user when the photo library is accessed for the first time</string>
```
* run
`react-native run-ios`

davebro, 2019-08-06:
@takitus Thanks for the install instructions. I set up a test project real quick and followed your instructions exactly, then tried referencing `RNCamera.checkVideoAuthorizationStatus()` in App.js, which gave me the same error referenced above.

[image]

Any suggestions? Thanks for your help.

takitus, 2019-08-06:
You shouldn't have to call that function manually. 

Try copy pasting the app.js from the basic example to the new test project you just made and running it. It should run, function, and pop up all necessary permission requests:

https://github.com/react-native-community/react-native-camera/blob/master/examples/basic/App.js

takitus, 2019-08-06:
Also, if you need to re-check permissions, you can use refreshAuthorizationStatus()

https://react-native-community.github.io/react-native-camera/docs/api#refreshauthorizationstatus

davebro, 2019-08-07:
@takitus Thank you for the info, that makes sense. One last question, if I wanted to pre-check for camera permissions on iOS before showing the official permissions modal, for instance to show our own modal first, is using _react-native-community/react-native-permissions_ the accepted way of doing that for iOS from this library's standpoint? Or is that what `refreshAuthorizationStatus` is for? Not seeing any examples of how that function works. I know we have _PermissionsAndroid_ on Android from react-native, but I'm wondering specifically about iOS.

That was the whole point of using `checkVideoAuthorizationStatus`, which is obviously now deprecated :)

closed 2019-11-03 by stale[bot]

williamgoodhew, 2020-01-15:
I am getting the same error
`TypeError: CameraManager.checkVideoAuthorizationStatus is not a function`

I am using React-Native: 0.61.5
react-native-camera: 3.15.1

Gerrist, 2020-08-26:
Are there any updates regarding this issue? Like others, I also successfully manually installed everything for IOS, but I still get the 'CameraManager.checkVideoAuthorizationStatus is not a function' error.

wxsms, 2020-09-04:
Manually install fixed this issue.

ref: #2275, #2248