react-native-camera/react-native-camera #2414: Could not find play-services-vision Android

brayanAP, 2019-08-06:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

rezaarifian, 2019-08-06:
same ,  I cannot release APK but debug success

any solution about play-services-vision ?

qimingweng, 2019-08-06:
Same issue here

MateusAndrade, 2019-08-11:
@brayanAP you can share your `android/app/build.gradle` and your `react-native` and `react-native-camera` version?

dlombardi, 2019-08-13:
same issue here

badver, 2019-08-26:
According to [docs](https://react-native-community.github.io/react-native-camera/docs/installation): 
Just add `googlePlayServicesVisionVersion = "<YOUR_VERSION_HERE>"` in `android/build.gradle`

play-services-vision versions you can found here: https://mvnrepository.com/artifact/com.google.android.gms/play-services-vision 

For my app I use these values:
```
ext {
        buildToolsVersion = "28.0.3"
        minSdkVersion = 16
        compileSdkVersion = 28
        targetSdkVersion = 28
        supportLibVersion = "28.0.0"
        googlePlayServicesVersion = "16.1.0"
        googlePlayServicesVisionVersion = "16.2.0"
    }
```

closed 2019-11-01 by stale[bot]

alexpoon2000, 2020-09-28:
I added "googlePlayServicesVisionVersion" and I can make the Android build now. However, the app crashed when I open it from Android device.