react-native-camera/react-native-camera #2415: Block Phone calls & Notification alerts while recording

athondapu, 2019-08-06:
# Question

**Ask your Question**
Hi I am very new to the react-native development. I am using this library and able to record the video and capture the photo. But I have a question is there anyway to block the phone calls and notifications while recording the video.? I searched in the web but I couldn't find any solution. Can any one please suggest or shed some light on how to block phone calls and notifications while recording the video.?

marcosrdz, 2019-08-06:
When it comes to iOS, this is not possible. I'm not sure if Android has special flags for it but I think is unlikely.

athondapu, 2019-08-15:
Thanks @marcosrdz for your reply. Is there anyway to block the calls & notifications in react-native.? If yes, can you please point me to the right direction.?

marcosrdz, 2019-08-15:
There seem to be ways to detect a call, but no way to block it:

https://www.npmjs.com/package/react-native-call-detection

Perhaps you could use the notification to manually pause recording and then continue once you get a notification of a call being complete.

closed 2019-10-21 by stale[bot]