react-native-camera/react-native-camera #2417: Barcode PDF417 in iOS is not working with Firebase MLKit

andresmtz98, 2019-08-06:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- react-native-camera: 3.0.1
- react-native: 0.60.4
- react: 16.8.6
- react-navigation: 3.11.0

**Description/Current Behaviour**

`onGoogleVisionBarcodesDetected` is not detecting barcode PDF417. That event is fired every second but the `barcodes` array is always empty :/.
I set a breakpoint but never is fired inside of `for` clause.

**Expected Behaviour**

Barcode PDF417 would be detected.

**Steps to Reproduce**

- Create a blank project
- Implement MLKit variant and use `onGoogleVisionBarcodesDetected`

**Additionals**

[Screenshot 1](https://user-images.githubusercontent.com/18200380/62581144-dff78f00-b86d-11e9-8a6e-1af23a6bf5aa.png)

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

MateusAndrade, 2019-08-11:
You tried using the `onBarCodeRead`? Also, if you can provide a example barcode that you are trying to scan, i can test it on a application that i worked. :smile:

andresmtz98, 2019-08-12:
Related to [barcode not recognized](https://github.com/firebase/firebase-ios-sdk/issues/2823).
I've tested with some barcodes so this problem is due ml-kit. It works fine with a couple of different barcodes but anothers can't be read, neither pdf417 from Colombian ID.

andresmtz98, 2019-08-12:
> You tried using the `onBarCodeRead`? Also, if you can provide a example barcode that you are trying to scan, i can test it on a application that i worked. 😄
> 
> Btw, this happens on Android?

Android works fine, that's my question why iOS doesn't works fine :/ because i think the ml model is the same xd.

wellington-cunha, 2019-08-16:
I solved my problem using  "react-native-camera": "1.2.0"

eactisgrosso, 2019-10-23:
Same here, working OK on Android but no luck in iOS.

brunocamposiws, 2019-12-23:
This is not fixed yeeeeeeeet

edritech93, 2020-01-02:
same for me

closed 2020-03-09 by stale[bot]

dayron9110, 2020-11-24:
I can't read Colombian id with iOS sdk... But in Android I can...