react-native-camera/react-native-camera #2420: autoFocusPointOfInterest not kept while capturing photo

cristianoccazinsp, 2019-08-08:
# Bug Report

**To Do First**

- [ ] Did you try latest release? Yes
- [ ] Did you try master? Yes
- [ ] Did you look for existing matching issues? Yes

**Platforms**

Android

**Versions**

- Android: 8 and 9
- react-native-camera: 3.1.0
- react-native: 0.60.4
- react: 16.86

**Description/Current Behaviour**

autoFocusPointOfInterest on Android correctly changes the camera focus while previewing. However, after calling ```takePictureAsync``` the focus is lost, and the camera captures the picture with a completely different focus. Lastly, the auto focus coordinates seem to reset very quickly (3 seconds max)

I suspect the 3 seconds focus reset comes from here: https://github.com/Craigtut/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L782

**Expected Behaviour**

Focus should be kept for longer, until the camera has moved enough like to trigger a new focus event. Capturing a photo should not attempt to re-focus.

cristianoccazinsp, 2019-08-08:
Setting DELAY_MILLIS_BEFORE_RESETTING_FOCUS to 10 seconds instead of 3 yields better results since it gives enough time for the camera to capture... I wonder if this could be made configurable. Otherwise, have the focus reset not to happen automatically unless the camera captured data changes, just like the native one.

sibelius, 2019-08-08:
We can make this customizable

cristianoccazinsp, 2019-08-08:
On my pixel 2 at least, I don't even see a reason for this to be needed at all. Focus is reset automatically by the camera even without this delay. Wish I could get some insight from the person that added that since I know every android device is it's own world and he probably had a very good reason for this.

cristianoccazinsp, 2019-08-10:
Capturing a photo also clears focus most likely due to this line: https://github.com/Craigtut/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L425

cristianoccazinsp, 2019-08-10:
Update: 
If we remove the call to autoFocus from https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L453  the camera becomes way faster and it also fixes https://github.com/react-native-community/react-native-camera/issues/2421

It also makes autoFocusPointOfInterest useful since you can capture with a custom focus now.

I wonder what are the consequences of this.

mattprivman, 2019-08-10:
@cristianoccazinsp Just found the same thing a few minutes ago. 

```
    @Override
    void takePicture(final ReadableMap options) {
        if (!isCameraOpened()) {
            throw new IllegalStateException(
                    "Camera is not ready. Call start() before takePicture().");
        }
        if (!mIsPreviewActive) {
            throw new IllegalStateException("Preview is paused - resume it before taking a picture.");
        }
        if (getAutoFocus()) {
            mCamera.cancelAutoFocus();
//            mCamera.autoFocus(new Camera.AutoFocusCallback() {
//                @Override
//                public void onAutoFocus(boolean success, Camera camera) {
                    takePictureInternal(options);
//                }
//            });
        } else {
            takePictureInternal(options);
        }
    }
```

I imagine it's what's doing the auto focusing when it's just set to auto, but needs testing to confirm. If that's the case when can set a conditional for when there's focus areas and metering areas set to simply not call that method.

Maybe someone more knowledgeable can chime in.

cristianoccazinsp, 2019-08-10:
I'm testing and just calling ```takePictureInternal(options);``` without any auto focus request is more than enough for the capture method, regardless of the autoFocusPointOfInterest property. It also fixes https://github.com/react-native-community/react-native-camera/issues/2421 (which has some odd bug and never fires the onAutoFocus event, ending up with the camera getting stuck).

I'm testing on more devices and with other people as well, but I can't seem to find any issue with this. If auto focus is set, you will most likely call the capture method only when you have a good enough focus, so calling this auto focus method is redundant in my opinion.

Doing this change also makes the camera look more like the IOS version which captures w/e is on the screen without trying to focus again.

mattprivman, 2019-08-10:
Not seeing any repercussions either so far. It's actually much nicer since it works closer to how the Native Camera works. It takes a picture of what the preview looks like rather than making any adjustments. 

Tracked down the git blame down to here: https://github.com/google/cameraview/blame/68947cc1643e7434250e099f38f346eae9c46339/library/src/main/api14/com/google/android/cameraview/Camera1.java#L222

cristianoccazinsp, 2019-08-10:
A bit unrelated, but why is the library using Camera1? Looking at the docs, Camera1 is deprecated. I couldn't find how to test the new Camera2 API, but it would be nice to know if it has the same issue: https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera2.java#L494

mattprivman, 2019-08-10:
It's undocumented but you have to set <RNCamera useCamera2Api="true"> for it to use Camera2. From experience, the Camera1 API seems to be more stable on react-native-camera.

cristianoccazinsp, 2019-08-10:
So doing the above change to just call ```takePictureInternal(options)``` without the autoFocus call, and setting ```DELAY_MILLIS_BEFORE_RESETTING_FOCUS``` to 10 seconds instead of 3, makes tap to focus and capture much much better, and the camera becomes closer to what the native one is.

Tested on pixel 2 and moto G5 for now. I wonder if a PR for this would be accepted, or do we need to demonstrate it won't break anything?

mattprivman, 2019-08-10:
Also tested on a Galaxy Tab Active 1 (Android 5.1.1) and a Galaxy Tab Active 2 (Android 9). 

A PR would be a good ground for discussion to see if it breaks anything.

cristianoccazinsp, 2019-08-14:
I wonder if we can also remove the calls to cancelAutoFocus that are here:
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L841
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L857
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L454
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L515
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L828

And the reset auto focus altogether that's called on a timeout after setting the point of interest:
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L801
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L821

Doing so would make the camera match closer what the native one does... At least on the moto G5 the focus is not reset automatically after capturing or moving the phone. The Pixel 2 seems to reset the focus area after large movements, but I guess we can't really detect that with the Camera API. Not doing it automatically still seems better than using a timeout to reset it on purpose.

cristianoccazinsp, 2019-08-14:
Changes here: https://github.com/react-native-community/react-native-camera/pull/2427

n1ru4l, 2019-08-17:
:tada: This issue has been resolved in version 3.1.2 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v3.1.2)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

cristianoccazinsp, 2020-08-11:
I have recently noticed this issue again on Android devices using Camera1.  I can't tell what changed, but taking a picture with `autoFocusPointOfInterest` clears the adjusted focus/white balance right after capturing a photo.  Nothing seems to have changed in the past few months that may change this...

cristianoccazinsp, 2020-08-11:
The behavior seems different now though. The photo is taken with the
previews focus and exposure but it resets right after it captures.  Testing
an older version on the same phone behaves the same, must have been a phone
update.

commit 34075331 (2019-08-17): fix(android): This update does two things in order to fix #2420 and #2421. (#2427)

First, it updates the take picture call to capture right away instead of trying to focus again. This will match closer to what the native camera does, and also what the Camera2 does.

Secondly, it removes all calls to resetFocus after capturing and after a given timeout. This was causing issues when using autoFocusPointOfInterest, and it really didn't make sense to use an internal timeout to reset the focus. If anything, the user should do this from the JS side. A new method could also be exposed to do this.

L [... cut]