react-native-camera/react-native-camera #2421: takePictureAsync does not return if focus changed

cristianoccazinsp, 2019-08-09:
# Bug Report

**To Do First**

- [ ] Did you try latest release? Yes
- [ ] Did you try master? Yes
- [ ] Did you look for existing matching issues? Yes

**Platforms**

Android

**Versions**

- Android: 8 and 9
- react-native-camera: 3.1.0
- react-native: 0.60.4
- react: 16.8.6
  

**Description/Current Behaviour**
If I call setState to change ```autoFocusPointOfInterest``` while the camera is trying to focus to capture after a call too ```takePictureAsync```, the method never resolves and neither rejects, it basically becomes stuck.

**Expected Behaviour**
takePictureAsync should always resolve or reject.

**Steps to Reproduce**

The code to capture looks like:
```
 try{
        data = await this.camera.takePictureAsync(options);
      }
catch(err){
        Alert.alert("Error", "Failed to take picture: " + err.message);
}
```

if a setState is called to update the autoFocusPointOfInterest RNCamera property before the above method resolves, it will never resolve. There are no errors on the android console neither.

cristianoccazinsp, 2019-08-10:
Update: If flash is turned on, flash/light also gets stuck until the camera component is unmounted. Tested on both Pixel 2 and Moto G5. This makes tap to focus nearly impossible to use on Android.

cristianoccazinsp, 2019-08-10:
Related to this: https://github.com/react-native-community/react-native-camera/issues/2420

If we remove the call to autoFocus before capturing all issues go away.
https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L453

cristianoccazinsp, 2019-08-10:
In addition to the above, if we don't remove the autoFocus call, this is most likely related to the additional calls to autoFocus and cancel/reset auto focus. The capture event is most likely lost when this happens and the promise is never resolved. Following https://github.com/react-native-community/react-native-camera/issues/2420, we most likely want to just get rid of the auto focus call (perhaps use a setting for it), which would solve both issues.

cristianoccazinsp, 2019-08-14:
Related changes: https://github.com/react-native-community/react-native-camera/pull/2427

n1ru4l, 2019-08-19:
Resolved in #2427

closed 2019-08-19 by n1ru4l