react-native-camera/react-native-camera #2422: How to get left/right eye position

ysrarff, 2019-08-09:
# Question

**To Do First**
- [X] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [X] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [X] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Question**

Hi there,

I was wondering if there was a way to obtain the left and right eye position of the face from FaceDetector module. I'm currently passing some .png frames inside the module and was able to obtain bounds, yawAngle and rollAngle but no luck on the other parameters.  

**My Podfile**

> platform :ios, '9.3'
require_relative '../node_modules/@react-native-community/cli-platform-ios/native_modules'
...
  pod 'Firebase/Core'
  pod 'GoogleMobileVision/FaceDetector'
  pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
    'FaceDetectorMLKit',
]
  pod 'react-native-ffmpeg', :podspec => '../node_modules/react-native-ffmpeg/ios/react-native-ffmpeg.podspec'
  pod 'react-native-video/VideoCaching', :path => '../node_modules/react-native-video/react-native-video.podspec'
  pod 'RNFS', :path => '../node_modules/react-native-fs'
...

**JS Code**

    import {RNCamera, FaceDetector} from 'react-native-camera';
    ...
    getFaces = async (imageFiles) => {
        const options = {
            mode: FaceDetector.Constants.Mode.accurate,
            detectLandmarks: FaceDetector.Constants.Landmarks.all,
            runClassifications: FaceDetector.Constants.Classifications.all
        }

        await FaceDetector.detectFacesAsync(imageFiles, options).then(faces => {
            console.log(faces);
            if (faces.length === 1) {
    
                const dateRef = new Date();
                const hour = dateRef.getUTCHours();
                const minute = dateRef.getUTCMinutes();
                const seconds = dateRef.getUTCSeconds();
                const milliseconds = dateRef.getUTCMilliseconds();
    
                const currentTime = hour + ':' + minute + ':' + seconds + ':' + milliseconds
    
                const rightEye = faces[0].rightEyeOpenProbability;
                const leftEye = faces[0].leftEyeOpenProbability;
                const blinkProb = (rightEye + leftEye) / 2;
    
                this.setState({
                    currentTime: currentTime
                });
    
                if (blinkProb >= 0.8) {
                    this.setState({ probability: blinkProb });
                    console.log('Eye is open at ' + this.state.currentTime + ' with a probability of ' + this.state.blinkProb);
                }
                else if (blinkProb <= 0.3) {
                    console.log('Eye is closed at ' + this.state.currentTime + ' with a probability of ' + this.state.blinkProb);
                    this.setState({ probability: blinkProb });
                }
            } else {
                // console.log("no face ");
            }
        }).catch(err => { console.log(err.message) });
    }

**Output from console.log(faces)**

[image]

If anyone could suggest a possible solution for me to try out that'd be great!

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

MateusAndrade, 2019-08-11:
Hi @ysrarff , you gave a look at the example app? Also, you can give a look at: https://firebase.google.com/docs/ml-kit/detect-faces. 

Please, let us know if you found anything that can help you.

ysrarff, 2019-08-15:
Closing this issue because I eventually got it working by using Firebase ML Vision package directly and bare RNCamera modules. Thanks @MateusAndrade !

closed 2019-08-15 by ysrarff