react-native-camera/react-native-camera #2424: Reading multiple barcodes at once on Android

iwashi1t, 2019-08-13:
# Feature-Request

**Describe the Feature**

Allow multiple barcodes to be read at once on Android (as with iOS).

Currently, for example, if there are three barcodes in the screen,  `onBarCodeRead` is called three times on iOS, but it seems to be called only once on Android.

**Possible Implementations**

How about using zxing's `GenericMultipleBarcodeReader.decodeMultiple` instead of `MultiFormatReader.decodeWithState`?

**Related Issues**

* #846
* #1593

yrichard, 2021-02-09:
Oh man, I really wish this was fixed...

yrichard, 2021-02-10:
Hi @MateusAndrade, any chance we can get that one re-opened? Or was there some specific limitations that prevent this from happen. Thank you :)

MateusAndrade, 2021-02-10:
Hi @yrichard, we can re-open that and tag this as a feature request, btw maybe this would be a nice increment at v43. What do you think @MarcoScabbiolo ?

MarcoScabbiolo, 2021-02-10:
I think scanning an entire frame for all the found barcodes and emitting all of them in a single event is something we can get done.

In v4 the scanning stops as soon as event is fired, and the event comes with a callback to resume the scanning. That's at least a starting point, and should prevent the du/triplicated events in iOS.

I agree this should go to v4.

yrichard, 2021-02-10:
Is it me or you guys are rocking hard?

closed 2021-04-21 by stale[bot]