react-native-camera/react-native-camera #2426: autoFocus stops working after pauseCapture / resumePreview

Kezmo, 2019-08-14:
- [ yes] Did you try latest release?
- [ yes] Did you try master?
- [ yes] Did you look for existing matching issues?

**Platforms**
Android
tried on a few samsungs, s8, s9, pixel.

**Description/Current Behaviour**
Calling pauseAfterCapture/pausePreview and then resumePreview locks the focus of the camera and autoFocus stops working.

**Expected Behaviour**
resumePreview should resume and keep auto focusing, allowing you to continue taking pictures at various distances.

**Steps to Reproduce**
pauseAfterCapture: true. in camera settings.
resumePreview(). In the takepictureasync data section.
autoFocus doesn't work anymore.

**Additionals**
I'm kinda new to all this stuff so will try help if i can. If there's something I've missed in the documentation and this is really easy to resolve then let me know.

cristianoccazinsp, 2019-08-14:
Can you see if the changes made in here fix this issue? Auto focus has some issues on the Camera1 implementation... https://github.com/react-native-community/react-native-camera/pull/2427

You don't need to install from it, you might as well modify the Camera1.java file right from your node_modules directory and test it. 
I doubt it will help with this, but the code had some cancel auto focus calls after capturing, which might be related to this.

As a work around you might want to play around with the autoFocusPointOfInterest and/or implement a tap to focus with it.

Kezmo, 2019-08-15:
Unfortunately this didn't fix the issue :(

cristianoccazinsp, 2019-10-14:
Hey @Kezmo , is this still an issue for you with the latest version? I've just tested on a Moto G5, and this doesn't seem to be an issue for me.

I'm using the `pauseAfterCapture` option for takePictureAsync, and then after capturing, calling resumePreview with a 3 seconds delay just to test this. Focus seems to be working fine all the time. Tap to focus also works fine.

JakubPod, 2019-12-13:
This still seems to be an issue. Tested on Samsung Galaxy J5 2017 today. Autofocus stops to work after recording a video and resuming a preview. Same with pauseAfterCapture.

cristianoccazinsp, 2019-12-13:
Can you post your camera component? Are you able to debug on with android studio / logcat to see if there's something else going on?

JakubPod, 2019-12-17:
There are no errors or anything out of ordinary in the logs. Furthermore, autofocus stops working after resuming preview and starts to works again once another recording starts.

QUICK FIX that worked for me, Camera1.java:
`    @Override
    public void resumePreview() {
        -startCameraPreview();
        +updateSurface();
    }`

cristianoccazinsp, 2019-12-17:
Does flash (e.g., torch), or zoom also break when preview is paused?

Rather than calling `updateSurface` which (as you said) is expensive and does a lot of redundant operations, you could try just refreshing the camera's parameters like ( I know this fixed flash while recording):

```
try{
                    mCamera.setParameters(mCameraParameters);
                } catch (Exception e) {
                    Log.e("CAMERA_1::", "Resume preview setParameters failed", e);
                }
```

Alternatively, you could also try calling `adjustCameraParameters();` instead.

Is that something you can try?  If that works, it might be a good idea to send a PR with the fix.

JakubPod, 2019-12-17:
Sure, I'll try your suggestions.

fellmann, 2020-02-19:
@JakubPod Did you try those suggestions? Could you provide a PR?

closed 2020-04-26 by stale[bot]