react-native-camera/react-native-camera #2428: On iOS it render a black screen and crash

wellington-cunha, 2019-08-15:
Hello everybody I am doing exactly what is the documentations.

My Podfile

platform :ios, '9.0'
require_relative '../node_modules/@react-native-community/cli-platform-ios/native_modules'

Sometimes it render the component on my phone and keep hanging on black screen and the snap buttom has no action and screen keep frozen. 

Sometimes it render the same black screen a and the app crashes 

target 'mobileapp' do
  # Pods for mobileapp
  pod 'React', :path => '../node_modules/react-native/'
  pod 'React-Core', :path => '../node_modules/react-native/React'
  pod 'React-DevSupport', :path => '../node_modules/react-native/React'
  pod 'React-fishhook', :path => '../node_modules/react-native/Libraries/fishhook'
  pod 'React-RCTActionSheet', :path => '../node_modules/react-native/Libraries/ActionSheetIOS'
  pod 'React-RCTAnimation', :path => '../node_modules/react-native/Libraries/NativeAnimation'
  pod 'React-RCTBlob', :path => '../node_modules/react-native/Libraries/Blob'
  pod 'React-RCTImage', :path => '../node_modules/react-native/Libraries/Image'
  pod 'React-RCTLinking', :path => '../node_modules/react-native/Libraries/LinkingIOS'
  pod 'React-RCTNetwork', :path => '../node_modules/react-native/Libraries/Network'
  pod 'React-RCTSettings', :path => '../node_modules/react-native/Libraries/Settings'
  pod 'React-RCTText', :path => '../node_modules/react-native/Libraries/Text'
  pod 'React-RCTVibration', :path => '../node_modules/react-native/Libraries/Vibration'
  pod 'React-RCTWebSocket', :path => '../node_modules/react-native/Libraries/WebSocket'

  pod 'React-cxxreact', :path => '../node_modules/react-native/ReactCommon/cxxreact'
  pod 'React-jsi', :path => '../node_modules/react-native/ReactCommon/jsi'
  pod 'React-jsiexecutor', :path => '../node_modules/react-native/ReactCommon/jsiexecutor'
  pod 'React-jsinspector', :path => '../node_modules/react-native/ReactCommon/jsinspector'
  pod 'yoga', :path => '../node_modules/react-native/ReactCommon/yoga'

  pod 'DoubleConversion', :podspec => '../node_modules/react-native/third-party-podspecs/DoubleConversion.podspec'
  pod 'glog', :podspec => '../node_modules/react-native/third-party-podspecs/glog.podspec'
  pod 'Folly', :podspec => '../node_modules/react-native/third-party-podspecs/Folly.podspec'
  pod 'RNCAsyncStorage', :path => '../node_modules/@react-native-community/async-storage'
  pod 'RNGestureHandler', :path => '../node_modules/react-native-gesture-handler/RNGestureHandler.podspec'

  pod 'RNVectorIcons', :path => '../node_modules/react-native-vector-icons'

  pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
    'BarcodeDetectorMLKit'
  ]

  target 'mobileappTests' do
    inherit! :search_paths
    # Pods for testing
  end

  use_native_modules!
end

target 'mobileapp-tvOS' do
  # Pods for mobileapp-tvOS

  target 'mobileapp-tvOSTests' do
    inherit! :search_paths
    # Pods for testing
  end

end
My package.json
{
  "name": "mobileapp",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint ."
  },
  "dependencies": {
    "@react-native-community/async-storage": "^1.6.1",
    "axios": "^0.19.0",
    "js-base64": "^2.5.1",
    "react": "16.8.6",
    "react-native": "0.60.4",
    "react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera.git",
    "react-native-gesture-handler": "^1.3.0",
    "react-native-ionicons": "^4.6.1",
    "react-native-paper": "^2.16.0",
    "react-native-permissions": "^1.2.0",
    "react-native-qrcode-scanner": "^1.2.1",
    "react-native-vector-icons": "^6.6.0",
    "react-navigation": "^3.11.1",
    "react-redux": "^7.1.0",
    "redux": "^4.0.4",
    "socket.io-client": "^2.2.0"
  },
  "devDependencies": {
    "@babel/core": "^7.5.5",
    "@babel/runtime": "^7.5.5",
    "@react-native-community/eslint-config": "^0.0.5",
    "babel-jest": "^24.8.0",
    "eslint": "^6.1.0",
    "jest": "^24.8.0",
    "metro-react-native-babel-preset": "^0.55.0",
    "react-test-renderer": "16.8.6"
  },
  "jest": {
    "preset": "react-native"
  }
}

wellington-cunha, 2019-08-16:
I solved the problem user version 1.2.0 :)

closed 2019-08-16 by wellington-cunha

VamshiKyra, 2019-08-21:
I had same issue. How to fix this? @wellington-cunha

wellington-cunha, 2019-08-21:
@VamshiKyra I downgraded to "react-native-camera": "1.2.0" now every things is ok