react-native-camera/react-native-camera #2432: iOS onFacesDetected fires off without faces

ghensley, 2019-08-16:
# Bug Report

**To Do First**

- [x ] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- iOS: 12.4 (iPhone X)
- react-native-camera: 3.1.1
- react-native: 0.60.4
- react: 16.8.6

**Description/Current Behaviour**

onFacesDetected is returning empty face arrays (like once per second) regardless of what it's looking at. The ground returned []. A human face returns [].

**Expected Behaviour**

Populated face arrays.

**Steps to Reproduce**

```import React, { Component } from "react";
import { View } from "react-native";
import { RNCamera } from "react-native-camera";

let readFromCamera = true;
class FaceScanner extends Component {
  state = {
    canDetectFaces: false
  };

  componentDidMount() {
    readFromCamera = true;
  }

  onFacesDetected = cameraResponse => {
    console.warn(JSON.stringify(cameraResponse));
  };

  onFaceDetectionError = error => {
    console.warn(JSON.stringify(error));
  };

  render() {
    return (
      <View style={{ position: "absolute", top: 0, left: 0, right: 0, bottom: 0 }}>
        <RNCamera
          style={{
            flex: 1
          }}
          ref={cam => {
            this.camera = cam;
          }}
          isTrackingEnabled={true}
          onFacesDetected={this.onFacesDetected}
          onFaceDetectionError={this.onFaceDetectionError}
          type="back"
        />
      </View>
    );
  }
}

export default FaceScanner;
```

Looks a lot like https://github.com/react-native-community/react-native-camera/issues/2318 but note the react-native-camera and react-native version bumps

Esxiel, 2019-08-25:
Hey, still facing the same problem since a month ago here. I'm on the same hardware and changing the orientation of the app into landscape and it works.

ghensley, 2019-08-29:
Appears to be the same as this as well, which looks to be related to an issue that was "fixed".

https://github.com/react-native-community/react-native-camera/issues/2395

closed 2019-11-04 by stale[bot]