react-native-camera/react-native-camera #2433: Testing Barcode Detection

JohannesWi, 2019-08-18:
Hey Everyone,

I'm comparing some Barcode-Scanner-Librarys for a project at the university. 

For that, It would be amazing if I could test the barcode-detection feature of this react-native-camera inside a test case (such as jest). 

Is there any way how I could simulate this inside a test case?

I do appreciate every help!!

hladf, 2019-09-05:
just use the google... https://stackoverflow.com/questions/41811477/how-to-mock-react-native-camera-with-jest

hladf, 2019-09-05:
https://www.google.com/search?q=react+native+camera+test+with+jest&rlz=1C1GCEU_pt-BRBR821BR822&oq=react+native+camera+test+with+j&aqs=chrome.1.69i57j33l5.9647j0j7&sourceid=chrome&ie=UTF-8

enjoy

JohannesWi, 2019-09-09:
Thank you, for your response!
I already looked up those solutions.

Of cause, its possible to Mock the Camera but I couldn't find a way, how to test the Barcode-detection feature. Do you have any ideas?

closed 2019-11-17 by stale[bot]