react-native-camera/react-native-camera #2436: AppState doesn't fire at the same time as recordAsync()

nachumFreedman, 2019-08-19:
# Bug Report
The appstate doesnt fire and when we put the app in the background then come back, the entire app is frozen to the last "frame" including react-native-video component. 

This happens on **Android** Google Pixel 2XL **ONLY!!!** 
We tested on other non-stok android devices.

nachumFreedman, 2019-08-19:
This is only in components that also contain recordAsync, didnt test other camera functionalities

nachumFreedman, 2019-08-20:
we had to hack the library. thanks

sibelius, 2019-08-20:
how to solve ?

cristianoccazinsp, 2019-08-21:
I reported the same issue some time ago (https://github.com/react-native-community/react-native-camera/issues/2235). No fixes yet, how did you solve it?

nachumFreedman, 2019-08-29:
I'm not a java dev but my co worker did something to the library where we shutdown the camera instead of pausing it.... I'll follow up with more details after the weekend

cristianoccazinsp, 2019-10-28:
@nachumFreedman  have you tried this issue with the latest version? I haven't had issues with it anymore.  There's also a PR that adds even more improvements with background threads and better locks for race conditions that should further prevent this issue and speed up the camera in general (https://github.com/react-native-community/react-native-camera/pull/2560)

closed 2020-01-03 by stale[bot]