react-native-camera/react-native-camera #2437: [Android] Camera2 may be wrong in converting image from YUV_420_888 format to byte array in `onImageAvailable`

minhthong095, 2019-08-20:
# Bug Report

**To Do First**
- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Environment**
- react-native-camera: 3.1.2
- react-native: 0.60.4
- react: 16.8.6

**Description/Current Behaviour**

Below is implement from `onImageAvailable` function in `Camera2.java`.

```
    private final ImageReader.OnImageAvailableListener mOnImageAvailableListener
            = new ImageReader.OnImageAvailableListener() {

        @Override
        public void onImageAvailable(ImageReader reader) {
            try (Image image = reader.acquireNextImage()) {
                Image.Plane[] planes = image.getPlanes();
                if (planes.length > 0) {
                    ByteBuffer buffer = planes[0].getBuffer();
                    byte[] data = new byte[buffer.remaining()];
                    buffer.get(data);
                    if (image.getFormat() == ImageFormat.JPEG) {
                        // @TODO: implement deviceOrientation
                        mCallback.onPictureTaken(data, 0);
                    } else {
                        mCallback.onFramePreview(data, image.getWidth(), image.getHeight(), mDisplayOrientation);
                    }
                    image.close();
                }
            }
        }
    };
```

I think `planes[0]` only get the Y chanel from the `image` when enable scanning QR code. And still contains 2 others planes ( U chanel and V chanel ) haven't counted yet when the `image` format is set `YUV_420_888` ( scanning activated ).

So the `BarCodeScannerAsyncTask.java` will receive missing byte data to decode. If compare to `Camera1.java`, the `public void onPreviewFrame(byte[] data, Camera camera)` method get completed data ( full 3 chanels ) in `NV21` format, it turn out QR decoded process is correct.

**Steps to Reproduce**
Since the `useCamera2Api` and `onBarCodeRead` [#2046](https://github.com/react-native-community/react-native-camera/issues/2046) issue has not fixed yet so I implement Camera2 and QR Scanning by my self and suddently figure out.

To workaround the issuse I have to convert the `image` to byte array like bellow code, thanks to this [post](https://stackoverflow.com/a/40152147). Before pass it to ` BarcodeDetectorAsyncTask.java` and barcode detect task work again.
```
    private fun imageToI420(image: Image): ByteArray {
        val yBuffer = image.planes[0].buffer
        val uBuffer = image.planes[1].buffer
        val vBuffer = image.planes[2].buffer

        val ySize = yBuffer.remaining()
        val uSize = uBuffer.remaining()
        val vSize = vBuffer.remaining()

        val i420 = ByteArray(ySize + uSize + vSize)

        yBuffer.get(i420, 0, ySize)
        uBuffer.get(i420, ySize, uSize)
        vBuffer.get(i420, ySize + uSize, vSize)

        return i420
    }
```

quanlm271, 2020-09-03:
I have got the same issue.

quanlm271, 2020-09-21:
after a while of search.
you can update with below code.

` mCallback.onFramePreview(YUV420toNV21(image), image.getWidth(), image.getHeight(), mDisplayOrientation);`

`private static byte[] YUV420toNV21(Image image) {
        Rect crop = image.getCropRect();
        int format = image.getFormat();
        int width = crop.width();
        int height = crop.height();
        Image.Plane[] planes = image.getPlanes();
        byte[] data = new byte[width * height * ImageFormat.getBitsPerPixel(format) / 8];
        byte[] rowData = new byte[planes[0].getRowStride()];

        int channelOffset = 0;
        int outputStride = 1;
        for (int i = 0; i < planes.length; i++) {
            switch (i) {
                case 0:
                    channelOffset = 0;
                    outputStride = 1;
                    break;
                case 1:
                    channelOffset = width * height + 1;
                    outputStride = 2;
                    break;
                case 2:
                    channelOffset = width * height;
                    outputStride = 2;
                    break;
            }

            ByteBuffer buffer = planes[i].getBuffer();
            int rowStride = planes[i].getRowStride();
            int pixelStride = planes[i].getPixelStride();

            int shift = (i == 0) ? 0 : 1;
            int w = width >> shift;
            int h = height >> shift;
            buffer.position(rowStride * (crop.top >> shift) + pixelStride * (crop.left >> shift));
            for (int row = 0; row < h; row++) {
                int length;
                if (pixelStride == 1 && outputStride == 1) {
                    length = w;
                    buffer.get(data, channelOffset, length);
                    channelOffset += length;
                } else {
                    length = (w - 1) * pixelStride + 1;
                    buffer.get(rowData, 0, length);
                    for (int col = 0; col < w; col++) {
                        data[channelOffset] = rowData[col * pixelStride];
                        channelOffset += outputStride;
                    }
                }
                if (row < h - 1) {
                    buffer.position(buffer.position() + rowStride - length);
                }
            }
        }
        return data;
    }`

reference: https://stackoverflow.com/a/59320979

closed 2021-01-13 by stale[bot]