react-native-camera/react-native-camera #2442: TypeError: CameraManager.checkIfRecordAudioPermissionsAreDefined is not a function

jmogielnicki, 2019-08-23:
# Bug Report

**To Do First**

- [ x ] Did you try latest release?
- [ ] Did you try master?
- [ x ] Did you look for existing matching issues?

**Platforms**
Android

**Versions**
- Android: 9
- iOS:
- react-native-camera: ^3.3.0
- react-native: 0.59.10
- react: 16.8.3
  

**Description/Current Behaviour**

On a clean project (macOs, expo, unmanaged (bare), I implemented the react native camera example from the docs, and see this error: TypeError: CameraManager.checkIfRecordAudioPermissionsAreDefined is not a function.

df80d267-74a2-4a5a-b…-460ced580311:15560 Possible Unhandled Promise Rejection (id: 0):
TypeError: CameraManager.checkIfRecordAudioPermissionsAreDefined is not a function
TypeError: CameraManager.checkIfRecordAudioPermissionsAreDefined is not a function
    at requestPermissions$ (blob:http://localhost:8081/df80d267-74a2-4a5a-b02c-460ced580311:94005:61)
    at tryCatch (blob:http://localhost:8081/df80d267-74a2-4a5a-b02c-460ced580311:41707:19)
    at Generator.invoke [as _invoke] (blob:http://localhost:8081/df80d267-74a2-4a5a-b02c-460ced580311:41882:24)
    at Generator.prototype.<computed> [as next] (blob:http://localhost:8081/df80d267-74a2-4a5a-b02c-460ced580311:41750:23)
    at tryCatch (blob:http://localhost:8081/df80d267-74a2-4a5a-b02c-460ced580311:41707:19)
    at invoke (blob:http://localhost:8081/df80d267-74a2-4a5a-b02c-460ced580311:41783:22)
    at blob:http://localhost:8081/df80d267-74a2-4a5a-b02c-460ced580311:41793:15
    at tryCallOne (blob:http://localhost:8081/df80d267-74a2-4a5a-b02c-460ced580311:45423:14)
    at blob:http://localhost:8081/df80d267-74a2-4a5a-b02c-460ced580311:45524:17
    at blob:http://localhost:8081/df80d267-74a2-4a5a-b02c-460ced580311:46402:21

**Expected Behaviour**

Camera package should work.

**Steps to Reproduce**

- expo init project
- Implement example per https://react-native-community.github.io/react-native-camera/docs/rncamera
- npm run android

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

jmogielnicki, 2019-08-23:
Hmm, I just tried the same repro steps with version ^2.6.0 and I'm seeing the same error, so probably not a regression @sibelius 

It's literally a brand new repo with the only change below in app.js.  Really struggling to understand what could be going on.

`'use strict';
import React, { PureComponent } from 'react';
import { AppRegistry, StyleSheet, Text, TouchableOpacity, View } from 'react-native';
import { RNCamera } from 'react-native-camera';

export default class App extends PureComponent {
  render() {
    return (
      <View style={styles.container}>
        <RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          style={styles.preview}
          type={RNCamera.Constants.Type.back}
          flashMode={RNCamera.Constants.FlashMode.on}
          androidCameraPermissionOptions={{
            title: 'Permission to use camera',
            message: 'We need your permission to use your camera',
            buttonPositive: 'Ok',
            buttonNegative: 'Cancel',
          }}
          androidRecordAudioPermissionOptions={{
            title: 'Permission to use audio recording',
            message: 'We need your permission to use your audio',
            buttonPositive: 'Ok',
            buttonNegative: 'Cancel',
          }}
          onGoogleVisionBarcodesDetected={({ barcodes }) => {
            console.log(barcodes);
          }}
        />
        <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
          <TouchableOpacity onPress={this.takePicture.bind(this)} style={styles.capture}>
            <Text style={{ fontSize: 14 }}> SNAP </Text>
          </TouchableOpacity>
        </View>
      </View>
    );
  }

  takePicture = async() => {
    if (this.camera) {
      const options = { quality: 0.5, base64: true };
      const data = await this.camera.takePictureAsync(options);
      console.log(data.uri);
    }
  };
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    flexDirection: 'column',
    backgroundColor: 'black',
  },
  preview: {
    flex: 1,
    justifyContent: 'flex-end',
    alignItems: 'center',
  },
  capture: {
    flex: 0,
    backgroundColor: '#fff',
    borderRadius: 5,
    padding: 15,
    paddingHorizontal: 20,
    alignSelf: 'center',
    margin: 20,
  },
});`

sibelius, 2019-08-23:
Maybe your link sin wrong

durac, 2019-08-26:
Are you aware that expo projects don't support native code? (https://docs.expo.io/versions/latest/introduction/faq/#how-do-i-add-custom-native-code)
I think that's the reason for the error.

jmogielnicki, 2019-09-01:
@durac thanks, I wasn't aware of this.  I will try it with a non-expo app.

closed 2019-09-01 by jmogielnicki

ss910423, 2019-10-11:
@jmogielnicki Did you solve this issue?

jmogielnicki, 2019-10-20:
@ss910423 no, I never did

Sentoki, 2020-05-25:
Hello, i am new with react and mobile development, and i want just make sure, that i understood case right.

- I want to play with mobile development, i want make application with single button that makes a photo
- On official site "get started" offers me use an expo
- And module from most popular repo can't work with expo

Am i right?