react-native-camera/react-native-camera #2444: RNCamera doesn'

xfyre, 2019-08-24:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 5.1.1
- react-native-camera: 3.3.0
- react-native: 0.60.5
- react: 16.8.6
- react-navigation: 3.11.0

**Description/Current Behaviour**

Code for QR code scanning doesn't work on Android (`onBarCodeRead` handler never gets called).

The issue is fixed by adding `apect: 1` to component parameters like this:

```clojure
              [qrs/rn-camera {:style {:flex 1}
                              :ref            #(do
                                                 (reset! camera-ref %)
                                                 (js/console.log %))
                              :onBarCodeRead  #(println "--- Bar code collected:" %)
                              :captureAudio   false
                              :aspect         1}
```

The fact that `aspect` parameter is undocumented on https://react-native-community.github.io/react-native-camera/docs/rncamera is very confusing.

**Expected Behaviour**

`RNCamera` component should be able to scan barcodes without any additional tweaks.

**Steps to Reproduce**

Create a basic example application with `RNCamera` on React Native 0.60.5 and try to scan a bar code.

**Additionals**

This wasn't an issue before upgrading to 0.60.x - everything worked flawlessly on 0.59.10 and below.

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

CorvinSteindl, 2019-08-29:
Having the same problem. But `aspect={1}` did not worked for me.

t4jfun, 2019-09-04:
Did you try taping the qr code when it appears on screen?

xfyre, 2019-09-04:
@t4jfun Yes, but it didn't help. The only workaround that helped me was specifying `aspect = 1` in component parameters.

agrass, 2019-09-04:
Hi, I'm with problems to scan barcodes with RN 0.59.x when I use Google Vision. It opens the camera but then crash without any log in react-native. Any idea what could be the problem? Which version of Google Vision are you using??

lisong996, 2019-10-15:
Having the same problem. But aspect={1} did not worked for me too.  
"react": "16.8.6",
"react-native": "0.60.5",
"react-native-camera": "^3.7.2",

closed 2019-12-24 by stale[bot]

ZiyaDogramaci, 2020-05-02:
The problem still exists.
**Platforms**

Android

**Versions**

Android: 6.0
react-native-camera: 3.23.1
react-native: 0.61.5
react: 16.9.0
react-navigation: 4.2.2

with the following props

`
                    aspect={1}
                    barCodeTypes={[RNCamera.Constants.BarCodeType.qr, 'qr']}
`