react-native-camera/react-native-camera #2458: Face bound origin is different with real position of face.

ohjeyong, 2019-08-30:
# Bug Report

**To Do First**

- [Y] Did you try latest release?
- [Y] Did you try master?
- [Y] Did you look for existing matching issues?

**Platforms**
Android

**Versions**
3.3.0

- Android: 9.0
- iOS:
- react-native-camera: 3.3.0
- react-native: 0.60.5
- react:
  

**Description/Current Behaviour**

When face is detected, the `face.bounds.origin.x` value is jumped at the certain point.
For example, when I placed the face to the left edge of my phone, `face.bounds.origin.x` is around 0. When I move my face gradually to the right, `face.bounds.origin.x` is gradually bigger. However, at some certain point, `face.bounds.origin.x` jumped to bigger. Also, that certain point is depending on the distance from phone and face.
When I place face closer to the phone and move gradually from left to right, `face.bounds.origin.x` comes to 0 ... 200 gradually, and then right after around 200, it comes 270. If I move slightly left, then it comes to 200. Slightly move right, it comes to 270.
If distance between face and phone is bigger, 0... 180 gradually, and it jumped to 240 around.

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

nguyenthaohut, 2019-09-05:
I met the same issue here for front camera, back camera running ok

nguyenthaohut, 2019-09-06:
face bound is working well on portrait mode, on landscape mode it happen this issue

closed 2019-11-12 by stale[bot]

samslow, 2020-01-16:
i met same but i'm using front camera. and origin y is problem in my case

myyrs, 2020-07-26:
same issue here

samslow, 2020-07-26:
@myyrs Try to use 'absolute' for position
wrong position error is relating to another Component.