react-native-camera/react-native-camera #2460: Error: Got exception while taking picture

jmogielnicki, 2019-09-01:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- iOS: 12.4.1
- react-native-camera: 3.3.10 (also tried 2.11.0)
- react-native: 0.59.10 (also tried 0.60.5)
- react: 16.8.6
  

**Description/Current Behaviour**

When implementing the [example RNCamera](https://react-native-community.github.io/react-native-camera/docs/rncamera), screen is black before a photo is taken (does not show preview video). When "snap" button is pressed, error is thrown

`Error: Got exception while taking picture
    at createErrorFromErrorData (blob:http://localhost:8081/ca06d1b7-b8d5-4de7-a013-feaebc4043d5:2118:17)
    at blob:http://localhost:8081/ca06d1b7-b8d5-4de7-a013-feaebc4043d5:2070:27
    at MessageQueue.__invokeCallback (blob:http://localhost:8081/ca06d1b7-b8d5-4de7-a013-feaebc4043d5:2637:18)
    at blob:http://localhost:8081/ca06d1b7-b8d5-4de7-a013-feaebc4043d5:2368:18
    at MessageQueue.__guard (blob:http://localhost:8081/ca06d1b7-b8d5-4de7-a013-feaebc4043d5:2541:13)
    at MessageQueue.invokeCallbackAndReturnFlushedQueue (blob:http://localhost:8081/ca06d1b7-b8d5-4de7-a013-feaebc4043d5:2367:14)
    at http://localhost:8081/debugger-ui/debuggerWorker.js:80:58`

**Expected Behaviour**

Camera should show the preview video and should be able to take a photo.

**Steps to Reproduce**
- react-native init cameraApp --version 0.59.10
- npm install react-native-camera --save
- react-native link react-native-camera
- run cameraApp in xcode
- press "capture"

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

jmogielnicki, 2019-09-01:
I pushed the code to github in case this is helpful in debugging: https://github.com/jmogielnicki/broken-react-native-camera

jmogielnicki, 2019-09-10:
Anyone have ideas about this?  I'm surprised because I'm using the example in a clean react-native app.

jmogielnicki, 2019-09-12:
@sibelius this is actually iOS, not Android

ohjeyong, 2019-09-19:
Same issue on iOS

liuminghui2017, 2019-09-24:
I got this issue too, delete the code below then then example work.
`          onGoogleVisionBarcodesDetected={({ barcodes }) => {
            console.log(barcodes);
          }}
`

jmogielnicki, 2019-10-20:
@liuminghui2017 thanks!  This fixes the issue for me as well.

closed 2019-10-20 by jmogielnicki