react-native-camera/react-native-camera #2468: Android crashes and freezes under various scenarios

cristianoccazinsp, 2019-09-06:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 8.1, 9, 10
- iOS:
- react-native-camera: 3.3.3
- react-native: 0.60.5
- react: 16.9.0

**Description/Current Behaviour**

So far I've found a bunch of cases where android will just crash or freeze under some camera stress (fast pictures, fast videos, rotation, home button presses, etc.)

1:  Home button right before picture is taken: This leaves the camera unusable. Basically, the capture callback is never called/resolved.

2: Rotation while picture is being taken: Same as above

3: Rapidly requesting new pictures with captureAsync: Same as above

4: Rotation + home button + stop video calls: A bit random, but video will often crash when trying to call mMediaRecorder.stop(). See crash logs below.

for 1,2,3 the issue is always the same, the callback is never fired (https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L510). My first guess is all the calls to stop preview, update params, orientation, etc. makes this callback never fire. This makes the promise never resolve, and makes the camera pretty much stuck.
Some comments here as well: https://stackoverflow.com/questions/6200369/picturecallback-onpicturetaken-never-called

for 4, it looks like a few concurrent calls to stop (https://github.com/react-native-community/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L161)
Stop will be called on record stop calls, host pause, surface destroyed, etc... Adding a few try/catch might help.

I've been trying to improve this but with no luck. There doesn't seem to be an easy way to detect the case where the camera callback is not fired and no picture is taken in order to reject the promise. 
I've also tried adding a few locks here and there, but locks will block the RN bridge risk freezing the app as well.

Have anyone seen something similar? Just try to rotate the device while capturing and you will end up in some of these cases.

cristianoccazinsp, 2019-09-06:
Here are the stack traces:

```
java.lang.RuntimeException set display orientation failed 
    Camera.java:-2 android.hardware.Camera.setDisplayOrientation
    Camera1.java:597 com.google.android.cameraview.Camera1.setDisplayOrientation
    CameraView.java:126 com.google.android.cameraview.CameraView$1.onDisplayOrientationChanged
    DisplayOrientationDetector.java:114 com.google.android.cameraview.DisplayOrientationDetector.dispatchOnDisplayOrientationChanged
    DisplayOrientationDetector.java:90 com.google.android.cameraview.DisplayOrientationDetector$1.onOrientationChanged
    OrientationEventListener.java:143 android.view.OrientationEventListener$SensorEventListenerImpl.onSensorChanged
    SystemSensorManager.java:837 android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent
    MessageQueue.java:-2 android.os.MessageQueue.nativePollOnce
    MessageQueue.java:336 android.os.MessageQueue.next
    Looper.java:174 android.os.Looper.loop
    ActivityThread.java:7356 android.app.ActivityThread.main
    Method.java:-2 java.lang.reflect.Method.invoke
    RuntimeInit.java:492 com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run
    ZygoteInit.java:930 com.android.internal.os.ZygoteInit.main
```

```
Caused by: java.lang.RuntimeException stop failed. 
    MediaRecorder.java:-2 android.media.MediaRecorder.stop
    Camera1.java:165 com.google.android.cameraview.Camera1.stop
    CameraView.java:308 com.google.android.cameraview.CameraView.stop
    RNCameraView.java:496 org.reactnative.camera.RNCameraView.onHostPause
    ReactContext.java:236 com.facebook.react.bridge.ReactContext.onHostPause
    ReactInstanceManager.java:674 com.facebook.react.ReactInstanceManager.moveToBeforeResumeLifecycleState
    ReactInstanceManager.java:498 com.facebook.react.ReactInstanceManager.onHostPause
    ReactInstanceManager.java:517 com.facebook.react.ReactInstanceManager.onHostPause
    ReactActivityDelegate.java:99 com.facebook.react.ReactActivityDelegate.onPause
    ReactActivity.java:58 com.facebook.react.ReactActivity.onPause
    Activity.java:7978 android.app.Activity.performPause
    Instrumentation.java:1507 android.app.Instrumentation.callActivityOnPause
    ActivityThread.java:4493 android.app.ActivityThread.performPauseActivityIfNeeded
    ActivityThread.java:4454 android.app.ActivityThread.performPauseActivity
    ActivityThread.java:4406 android.app.ActivityThread.handlePauseActivity
    PauseActivityItem.java:46 android.app.servertransaction.PauseActivityItem.execute
    TransactionExecutor.java:176 android.app.servertransaction.TransactionExecutor.executeLifecycleState
    TransactionExecutor.java:97 android.app.servertransaction.TransactionExecutor.execute
    ActivityThread.java:2016 android.app.ActivityThread$H.handleMessage
    Handler.java:107 android.os.Handler.dispatchMessage
    Looper.java:214 android.os.Looper.loop
    ActivityThread.java:7356 android.app.ActivityThread.main
    Method.java:-2 java.lang.reflect.Method.invoke
    RuntimeInit.java:492 com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run
    ZygoteInit.java:930 com.android.internal.os.ZygoteInit.main
```

```
java.lang.IllegalStateException
    MediaRecorder.java:-2 android.media.MediaRecorder.stop
    Camera1.java:165 com.google.android.cameraview.Camera1.stop
    Camera1.java:139 com.google.android.cameraview.Camera1$1.onSurfaceDestroyed
    PreviewImpl.java:60 com.google.android.cameraview.PreviewImpl.dispatchSurfaceDestroyed
    TextureViewPreview.java:59 com.google.android.cameraview.TextureViewPreview$1.onSurfaceTextureDestroyed
    TextureView.java:259 android.view.TextureView.releaseSurfaceTexture
    TextureView.java:230 android.view.TextureView.onDetachedFromWindowInternal
    View.java:19623 android.view.View.dispatchDetachedFromWindow
    ViewGroup.java:3806 android.view.ViewGroup.dispatchDetachedFromWindow
    ViewGroup.java:3806 android.view.ViewGroup.dispatchDetachedFromWindow
    ViewGroup.java:3806 android.view.ViewGroup.dispatchDetachedFromWindow
    ViewGroup.java:3806 android.view.ViewGroup.dispatchDetachedFromWindow
    ViewGroup.java:3806 android.view.ViewGroup.dispatchDetachedFromWindow
    ViewGroup.java:3806 android.view.ViewGroup.dispatchDetachedFromWindow
    ViewGroup.java:3806 android.view.ViewGroup.dispatchDetachedFromWindow
    ViewGroup.java:5431 android.view.ViewGroup.removeViewInternal
    ViewGroup.java:5378 android.view.ViewGroup.removeViewAt
    ReactViewGroup.java:468 com.facebook.react.views.view.ReactViewGroup.removeViewAt
    ReactViewManager.java:353 com.facebook.react.views.view.ReactViewManager.removeViewAt
    ReactViewManager.java:39 com.facebook.react.views.view.ReactViewManager.removeViewAt
    NativeViewHierarchyManager.java:450 com.facebook.react.uimanager.NativeViewHierarchyManager.manageChildren
    UIViewOperationQueue.java:228 com.facebook.react.uimanager.UIViewOperationQueue$ManageChildrenOperation.execute
    UIViewOperationQueue.java:844 com.facebook.react.uimanager.UIViewOperationQueue$1.run
    UIViewOperationQueue.java:952 com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches
    UIViewOperationQueue.java:44 com.facebook.react.uimanager.UIViewOperationQueue.access$2200
    UIViewOperationQueue.java:1012 com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded
    GuardedFrameCallback.java:29 com.facebook.react.uimanager.GuardedFrameCallback.doFrame
[... cut ...]
    Choreographer.java:951 android.view.Choreographer$FrameDisplayEventReceiver.run
    Handler.java:883 android.os.Handler.handleCallback
    Handler.java:100 android.os.Handler.dispatchMessage
    Looper.java:214 android.os.Looper.loop
    ActivityThread.java:7356 android.app.ActivityThread.main
    Method.java:-2 java.lang.reflect.Method.invoke
    RuntimeInit.java:492 com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run
    ZygoteInit.java:930 com.android.internal.os.ZygoteInit.main

```

cristianoccazinsp, 2019-09-06:
Pull request here for discussion. I believe these changes improve most of the above cases and reduce crashes/ANRs considerably.

https://github.com/react-native-community/react-native-camera/pull/2471

MateusAndrade, 2019-09-09:
@cristianoccazinsp thanks for helping us. Since #2471 was merged, do you think we can close this issue?

cristianoccazinsp, 2019-09-09:
@MateusAndrade I think so. I will open another issue/PR if I find new crashes or other cases I didn't handle with the above PR. Does that sound right?

Also, as I've commented in the PR, pressing the home button / locking the phone (or anything that stops the camera) will still end up with an unresolved promise if there was a photo capture in progress. I wasn't able to improve that case in the PR, and fixing it probably means a lot of work and might not even be worth it.

closed 2019-09-09 by sibelius