react-native-camera/react-native-camera #2470: Android BLU phones producing .3gp videos with .mp4 filename extensions

NightWarrior, 2019-09-06:
# Bug Report

**Platforms**
React-Native, Android 

**Versions**

- Android: Multiple
- react-native-camera: 2.11
- react-native: 0.59.9
- react: 16.8.3

**Description/Current Behaviour**

App records and uploads videos to a server which can be played by another entity and is expecting a .mp4 format. Issue was found that BLU android phones seem to be producing files with 3gp signature at start if opened in a text editor. 'BLU NEO X LTE' very certainly does give this issue. 

**Expected Behaviour**

Video should be in .mp4 file format or, Correct file extension for the file format and an output to denote which format the video output is in.

**Steps to Reproduce**

use 'recordAsync' with a BLU phone and check file at given path.
Tried both 4:3 and 480p, both gave the .3gp output videos

**Additionals**

If anyone can suggest, what would be the most efficient way to obtain a .mp4 if we cant go around having to convert the .3gp into .mp4

Also, i personally dont have access to a BLU device, but we have verified through a few BLU devices and this seems to be the case.

sanrodari, 2019-10-03:
Dump spec for the video that Taimoor(@NightWarrior) is referring to: https://www.get-metadata.com/result/7fe03ff4-8840-473c-8753-9accf6b2d95c

closed 2019-12-25 by stale[bot]

ziahm, 2020-06-04:
It is still an issue. When a video is recorded, it returns the recorded video path extension ".mp4" but its mime type is "video/3gpp"