react-native-camera/react-native-camera #2472: Camera default view is zoomed

kp2506, 2019-09-07:
Hi All,

I am using react-native-camera with one of my android app. I have tried with 3.3.0, 3.3.1, 3.3.2 the issue is the default view of camera is zoomed, I have compared it to default camera app in the phone.

I have tried setting zoom props to 0 (zero), but the behaviour is not changed.

Please help.

jqn, 2019-09-10:
I'm on 2.6.0 and the cameras match the view. I haven't upgraded to the latest version but I hope this helps. Also make sure you are setting the ratio properly for android

closed 2019-11-19 by stale[bot]

kulek1, 2019-11-21:
I have the same problem on Android with `3.8.0` and `3.11.1` version.
iOS works like a charm.

ksairi, 2019-11-28:
@kulek1 I'm using expo-camera, I'm having this same problem in iOS 13.2.3. Which iOS version are you on that works for you?

alyyousuf7, 2020-02-12:
ExpoCamera has the issue on iOS 13.3.1
I came here wishing that RNCamera doesn't have the same issue as ExpoCamera. 😞

sawndur, 2020-12-23:
Hey all, I think I found the issue. In RNCamera.m in the -initWithBridge function, 

 self.previewLayer.videoGravity = AVLayerVideoGravityResizeAspectFill; 

This is stretching out the aspect ratio giving a zoomed effect. You need to change it to AVLayerVideoGravityResizeAspect;