react-native-camera/react-native-camera #2473: videoBitrate option for iOS

xHeinrich, 2019-09-08:
# Feature-Request

**Describe the Feature**
Ideally the same as the android option.

**Possible Implementations**
I believe this can be implemented using the following key around here: https://github.com/react-native-community/react-native-camera/blob/master/ios/RN/RNCamera.m#L601 though I'm no iOS dev so i'm not 100%.

https://developer.apple.com/documentation/avfoundation/avvideoaveragebitratekey?language=objc
https://developer.apple.com/documentation/avfoundation/avcapturemoviefileoutput?language=objc

**Related Issues**
https://github.com/react-native-community/react-native-camera/issues/687
https://github.com/react-native-community/react-native-camera/pull/2055

xHeinrich, 2019-09-20:
I've hacked this into my own app with some hard coding. If anyone wants to improve upon this and submit a PR feel free to do so.

`    if (@available(iOS 10, *)) {
        AVVideoCodecType videoCodecType = options[@"codec"];
        NSString *videoBitrate = options[@"videoBitrate"];
        NSLog(@"VIDEO BITRATE %@", videoBitrate);
        [self.movieFileOutput setOutputSettings:@{
          AVVideoCodecKey:AVVideoCodecH264,
          AVVideoCompressionPropertiesKey:
              @{
                  AVVideoAverageBitRateKey:videoBitrate
              }
          } forConnection:connection];
    }
`

sibelius, 2019-09-24:
fixed here https://github.com/react-native-community/react-native-camera/pull/2504